CloverHackyColor/CloverBootloader #502: Reboot loop issue during update Monterey with legacy Clover 514x versions

jsl2000, 2022-02-28:
I have several old PC which need booting with legacy Clover or OpenCore.
This reboot loop issue only happened during update of Monterey with legacy Clover.
At present I can update Monterey with legacy OpenCore only, then it can be booted with legacy Clover.
My involved PC include the following:
MB: ASUS P5Q PRO | CPU: Intel D2C | GPU: GT-730
MB: ASUS P6TSE | CPU: Intel i7-920 | GPU: R9-290X
MB: ASUS M5A78L-M | CPU: AMD FX-6300 | GPU: RX-560
MB: GA-965P-DS4 | CPU: Intel D2C E8400 | GPU: GT-710
Hope it can be solved as soon as possible because I prefer Clover for these old PC.

jsl2000, 2022-03-01:
I always select Install for first reboot, but still got reboot loop issue.
Based upon my experiment at more than 10 Hackintoshs my conclusion:
1. It only occurred at legacy Clover because it created a defected partition which can not be installed completely by any Hackintoshs.
2. Using UEFI Clover or OpenCore created partition, then it can be installed successfully by any Hackintoshs including legacy Clover booting one.  
3. I hope this information may help you to debug this issue which bothered many Hackintosh user who need legacy Clover to install or update Big Sur and Monterey.

SergeySlice, 2022-03-13:
But look here
https://www.insanelymac.com/forum/topic/350590-pre-release-macos-monterey-123/?do=findComment&comment=2777994

SergeySlice, 2022-05-17:
See another user updated Monterey on legacy computer
https://www.insanelymac.com/forum/topic/348279-hp-dc7800sff-monterey/?do=findComment&comment=2778272

jsl2000, 2022-06-11:
I found there is no OpenRuntime.efi in his EFI/CLOVER/drivers/BIOS, how can it boot Monterey ?
There is OpenRuntime.efi in his EFI/CLOVER/drivers/UEFI only.

SergeySlice, 2022-06-21:
Legacy Clover not need OpenRuntime.efi.

ellaosx2007, 2022-06-23:
> Legacy Clover not need OpenRuntime.efi.

On my part, i still need it else no nvidia acceleration due to nvram.

SergeySlice, 2022-06-29:
With legacy Clover there is emulated nvram so Nvidia drivers work fine.

jsl2000, 2022-06-30:
Clover-5147 still had this reboot loop issue in all my legacy Clover booting hackintoshs.

ellaosx2007, 2022-07-03:
> With legacy Clover there is emulated nvram so Nvidia drivers work fine.

Sadly, Not for me

SergeySlice, 2023-10-24:
New Clover 5155 has better NVRAM support due to using advanced nvram2 utility. Consider to test it.

SergeySlice, 2023-10-24:
> Legacy Clover not need OpenRuntime.efi.

May be I am wrong and legacy Clover also uses OpenRuntime.efi.

SergeySlice, 2024-01-19:
Resolved

closed 2024-01-19 by SergeySlice