CloverHackyColor/CloverBootloader #504: cant get system updates

taqieallah, 2022-03-07:
hi .. why i cant get any system update on my lover efi while oc working good ? i enabled sip to no action happend

SergeySlice, 2022-03-07:
It depends on SMBIOS model and HWTarget.
But Clover new version does this automatically. Check if you have latest version.

taqieallah, 2022-03-07:
Which version ? im using mbp 16.4 as i have 10750h processor ... and version 5144 doesnt do it

SergeySlice, 2022-03-13:
Latest version is 5145.
What do you have in
~~~~
        <key>RtVariables</key>
        <dict>
            <key>HWTarget</key>
            <string>
~~~~
?
If you have no such key then Clover will automatically set J215AP
As well SIP you said enabled. Good.

SergeySlice, 2022-03-25:
Abandoned == resolved

closed 2022-03-25 by SergeySlice