CloverHackyColor/CloverBootloader #506: Incrimental updates

VineshPatel-keep-it, 2022-03-16:
Can't update Bigsur 11.5 to 11.6.5 with Clover EFI latest version. I had selected Boot macOs Install via Hackintosh HD Preboot on clover boot screen. And got boot loop. In OpenCore I was able to do Incremental updates.

SergeySlice, 2022-03-18:
I updated from BigSur 11.0 incrementally up to Monterey 12.3. Why not you?
First boot to Install, second and further to Preboot.

VineshPatel-keep-it, 2022-03-21:
> I updated from BigSur 11.0 incrementally up to Monterey 12.3. Why not you? First boot to Install, second and further to Preboot.

I tried but not worked. Can you send me your EFI with blank SMBIOS. So I can match with mine one and correct my EFI?

SergeySlice, 2022-03-21:
Why not this one? https://github.com/CloverHackyColor/CloverBootloader/releases/download/5145/CloverV2-5145.zip
I can send you my folder but it depends on hardware. Are you sure we have similar hardware?

VineshPatel-keep-it, 2022-03-21:
> Why not this one? https://github.com/CloverHackyColor/CloverBootloader/releases/download/5145/CloverV2-5145.zip
> 
> I can send you my folder but it depends on hardware. Are you sure we have similar hardware?

No sir mine hardware isn't the same but I can try to make my EFI from your EFI. So can you provide your EFI with blank SMBIOS sir?

VineshPatel-keep-it, 2022-03-23:
> Why not this one? https://github.com/CloverHackyColor/CloverBootloader/releases/download/5145/CloverV2-5145.zip I can send you my folder but it depends on hardware. Are you sure we have similar hardware?

Can you send me your EFI sir?

SergeySlice, 2022-03-25:
> > Why not this one? https://github.com/CloverHackyColor/CloverBootloader/releases/download/5145/CloverV2-5145.zip I can send you my folder but it depends on hardware. Are you sure we have similar hardware?
> 
> Can you send me your EFI sir?

OK, I give you my EFI folder with empty serial number. Don't forget to fill it.
But see carefully other settings. All these files depends on my hardware. It works for me not for you.
Sure this folder will not work for you as is.
[EFI.7z.zip](https://github.com/CloverHackyColor/CloverBootloader/files/8352371/EFI.7z.zip)
As well it contains my DSDT.aml. It is description of my hardware not your hardware. Don't use my DSDT!!!

VineshPatel-keep-it, 2022-03-26:
Brother I've made my EFI inspired by your EFI. Can give an eye on my EFI and check my hardware remotely? With teamviewer? Is my EFI well enough?

SergeySlice, 2022-04-13:
This is not a place to ask a help. The issue system is for developement.

closed 2022-04-13 by SergeySlice