CloverHackyColor/CloverBootloader #509: Feature request: chainloading a 64-bit (x64_64) UEFI from a 32-bit (x86/ia32) UEFI

volkertb, 2022-04-17:
I know what I'm proposing here isn't trivial, but please bear with me. (The first three following paragraphs are to provide context for those reading this who are not familiar with the phenomenon of 64-bit Intel systems shipping with 32-bit UEFIs. Just skip those if you're aware of the problem.)

So some years back, Intel released Bay Trail and Cherry Trail SoCs (used in small and low power form factors such as the Intel Compute Stick and some x86 tablets), and supplied OEMs with a 32-bit UEFI for them, even though the CPUs were perfectly capable of running 64-bit code. The reasoning behind this was that it would be more optimal to have them boot 32-bit versions of Windows, since such SoCs didn't ship with more than 2GB of RAM anyway, and running 64-bit versions of Windows would take up more RAM. This was of course short-sighted, because it didn't take into account the longevity of 32-bit x86 operating systems and the long-term application support for that architecture. (This decision also reeks a bit of "planned obsolescence" to me, but I digress.)

Indeed, Windows 8 and 10 on UEFI require a 32-bit UEFI for the 32-bit variants of the OS, and a 64-bit UEFI for the 64-bit variants. This cannot be mixed. However, the Linux kernel eventually gained support for "mixed mode UEFI", allowing a 64-bit Linux kernel to boot from a 32-bit UEFI. Some Linux distros, particularly Ubuntu and Fedora, support this out-of-the-box in their regular 64-bit installation media. [Chrome OS Flex](https://chromeenterprise.google/os/chromeosflex/) supports this as well. (I've tested this on a Bay Trail device with a 32-bit UEFI and it works, in 64-bit mode.) Notably however, Clear Linux* OS, developed and funded by Intel, [refuses to support this mixed mode](https://github.com/clearlinux/distribution/issues/262), even though Intel caused this mess in the first place.

Now, had the 32-bit UEFIs in these SoCs included a CSM to allow legacy BIOS compatibility, Clover could have been used as a workaround to boot 64-bit versions of Windows and Clear* Linux from such systems. Unfortunately, a CSM was not included in these implementations, so that option is not available.

In addition to these Intel SoCs, there were also some older generation Apple Mac products that had basically the same conundrum (a 64-bit capable CPU limited by a 32-bit EFI firmware).

Which brings me to this feature request: would it be feasible to port Clover to a 32-bit UEFI boot payload that launches its CloverEFI UEFI firmware emulation in 64-bit mode? So in effect chainloading (or bootstrapping) a 64-bit UEFI implementation from a 32-bit UEFI? The 64-bit Linux kernel is capable of booting into 64-bit mode from a 32-bit UEFI, using the aforementioned mixed mode support, so why couldn't Clover be made to use the same mechanism?

This would "liberate" such 64-bit-capable  platforms that would otherwise be stuck in 32-bit land, and extend their usefulness instead of these things ending up on a landfill. Sure, with 2GB RAM, their functionality would still be limited, but there are plenty of applications that these things could still be used for. The 32-bit x86 platform is basically end-of-life, and more and more applications are no longer being released for 32-bit, even though they don't require more than 2GB of RAM to work.

I'm interested in hearing what obstacles there would be towards such a solution.

Thank you for your consideration.

Some other related projects that might be useful to mention here:

- [macosxbootloader](https://github.com/Piker-Alpha/macosxbootloader), boot loader for older 64-bit Macs with 32-bit EFI firmware, to be able to boot newer 64-bit versions of Mac OS X, using the same mixed mode (thunking) mechanism
- A (currently still open) issue about properly supporting [(U)EFI Mixed Mode in systemd](https://github.com/systemd/systemd/issues/17056)

SergeySlice, 2022-04-19:
Yes, it is possible. Currently Clover has legacy boot mode and some years ago it could be 32-bit mode supporting 32-bit UEFI Runtime Functions. 
Anyway this is necrophilia and if someone want to restore 32bit abilities then welcome to be a developer yourself.

unrelentingtech, 2022-05-20:
Custom firmware based on coreboot+edk2 might be a good alternative for making these devices more useful. It's a way more invasive modification but it would permanently turn them into modern systems. But yeah, for quick testing especially, a non-invasive option would be better.

volkertb, 2022-06-12:
> Yes, it is possible. Currently Clover has legacy boot mode and some years ago it could be 32-bit mode supporting 32-bit UEFI Runtime Functions. Anyway this is necrophilia and if someone want to restore 32bit abilities then welcome to be a developer yourself.

@SergeySlice Sorry for the confusion, but that was not what I meant. I didn't mean restoring the 32bit abilities of Clover. I am asking for a 64-bit version of Clover that could be bootstrapped from a 32-bit UEFI on otherwise 64-bit systems, such as older Macs and some Intel SoCs such as Bay Trail. So basically a 64-bit UEFI being chain-loaded from the 32-bit UEFI that those systems were shipped with (and which lacks a CSM, making it impossible for Clover to be booted on those systems in Legacy BIOS mode). This would allow such systems to boot the 64-bit versions of most common operating systems without requiring any hacks or patches.

> Custom firmware based on coreboot+edk2 might be a good alternative for making these devices more useful. It's a way more invasive modification but it would permanently turn them into modern systems. But yeah, for quick testing especially, a non-invasive option would be better.

I agree, having Coreboot ported to such systems would be better still, and [someone inquired about that in this thread in the Intel community forums](https://community.intel.com/t5/Embedded-Intel-Atom-Processors/Baytrail-coreboot-support/m-p/189568), but that appears to have gone no-where, unfortunately.

SergeySlice, 2022-06-21:
If there is no CSM then it is impossible to run legacy Clover which can boot other OSes.
If there is 32bit UEFI then you can start Clover-32bit in UEFI mode. It is impossible to run Clover-64 with 32bit UEFI because Clover uses runtime functions like gBS->LoadImage() which is 32 bits here.
The only way is to use CSM.

SergeySlice, 2022-06-29:
Clover 32bit can boot macOS in 64bit mode. As well Linux+Grub itself know what to do. Windows too.

volkertb, 2022-06-30:
Okay, so there is a wat tot het 64-bit macOS tot boot from a 32-bit UEFI.

But "Windows too"? Only Windows ia32. Which is a legacy architecture at this point. Windows x64 still requires either 64-bit UEFI or legacy BIOS (CSM). And there are sone Linux distros, notably Clear Linux*, that don't support Grub and also require a 64-bit UEFI.

There is a use case for this.

rafi16jan, 2023-03-28:
> Okay, so there is a way for 64-bit macOS to boot from a 32-bit UEFI.
> 
> But "Windows too"? Only Windows ia32, which is a legacy architecture at this point. Windows x64 still requires either 64-bit UEFI or legacy BIOS (CSM). And there are some Linux distros, notably Clear Linux*, that don't support Grub and also require a 64-bit UEFI.
> 
> There is a use case for this.

@SergeySlice does clover support booting legacy boot OS (Windows 10 x64 legacy mode for example) from IA32 EFI?

NeoLizzard, 2023-04-25:
I would love to see it working too. This was pretty much my last night, trying to make my baytrail schenker element tablet go x32 uefi -> x64 uefi. I managed so far to force feed the the disk by using dism for putting Windows 11 x64 on the drive (yeah... madnass...) and then went with all loading hacks that I can come up with. However the limitation here was hit in my case when I hit winload.exe (since the call was still beeing made by ia32). I also tried to chainload from bootia32.efi into bootx64.efi - with quiet no success... My idea was to try to use grub with a grub config to do it - But so far no progress that is worth mention. It was just a couple of hours of try and error. Check easy2boot as a base, since they support a huge amount of boot processes for loading their iso stuff. Maybe you find something that I have overlooked. https://easy2boot.xyz/ (and yes... page look like shit... but the tool is good...)

Viking8, 2023-06-19:
I've been banging my head against a wall trying to get 64 bit Linux Mint to boot from internal storage on my Dell Venue 10 Pro 5055 tablet without success. I did try Debian 12 the day it was released and it actually booted, but man was it slow. The LM 21.1 live USB ran faster. Some things in Debian didn't work properly, but everything worked in Mint. At one point I thought clover might be my saviour, but it didn't work either, so I'd love to see clover adopt this function as well.

alcatraz-zz, 2024-01-11:
This thread was a great read. I'm also trying to boot win64 on ia32 efi.

PC-Doctor666, 2024-01-21:
> I know what I'm proposing here isn't trivial, but please bear with me. (The first three following paragraphs are to provide context for those reading this who are not familiar with the phenomenon of 64-bit Intel systems shipping with 32-bit UEFIs. Just skip those if you're aware of the problem.)
> 
> So some years back, Intel released Bay Trail and Cherry Trail SoCs (used in small and low power form factors such as the Intel Compute Stick and some x86 tablets), and supplied OEMs with a 32-bit UEFI for them, even though the CPUs were perfectly capable of running 64-bit code. The reasoning behind this was that it would be more optimal to have them boot 32-bit versions of Windows, since such SoCs didn't ship with more than 2GB of RAM anyway, and running 64-bit versions of Windows would take up more RAM. This was of course short-sighted, because it didn't take into account the limited remaining longevity of 32-bit x86 operating systems and the lack of further long-term application support for that architecture. (This decision also reeks a bit of "planned obsolescence" to me, but I digress.)
> 
> Indeed, Windows 8 and 10 on UEFI require a 32-bit UEFI for the 32-bit variants of the OS, and a 64-bit UEFI for the 64-bit variants. This cannot be mixed. However, the Linux kernel eventually gained support for "mixed mode UEFI", allowing a 64-bit Linux kernel to boot from a 32-bit UEFI. Some Linux distros, particularly Ubuntu and Fedora, support this out-of-the-box in their regular 64-bit installation media. [Chrome OS Flex](https://chromeenterprise.google/os/chromeosflex/) supports this as well. (I've tested this on a Bay Trail device with a 32-bit UEFI and it works, in 64-bit mode.) Notably however, Clear Linux* OS, developed and funded by Intel, [refuses to support this mixed mode](https://github.com/clearlinux/distribution/issues/262), even though Intel caused this mess in the first place.
> 
> Now, had the 32-bit UEFIs in these SoCs included a CSM to allow legacy BIOS compatibility, Clover could have been used as a workaround to boot 64-bit versions of Windows and Clear* Linux from such systems. Unfortunately, a CSM was not included in these implementations, so that option is not available.
> 
> In addition to these Intel SoCs, there were also some older generation Apple Mac products that had basically the same conundrum (a 64-bit capable CPU limited by a 32-bit EFI firmware).
> 
> Which brings me to this feature request: would it be feasible to port Clover to a 32-bit UEFI boot payload that launches its CloverEFI UEFI firmware emulation in 64-bit mode? So in effect chainloading (or bootstrapping) a 64-bit UEFI implementation from a 32-bit UEFI? The 64-bit Linux kernel is capable of booting into 64-bit mode from a 32-bit UEFI, using the aforementioned mixed mode support, so why couldn't Clover be made to use the same mechanism?
> 
> This would "liberate" such 64-bit-capable platforms that would otherwise be stuck in 32-bit land, and extend their usefulness instead of these things ending up on a landfill. Sure, with 2GB RAM, their functionality would still be limited, but there are plenty of applications that these things could still be used for. The 32-bit x86 platform is basically end-of-life, and more and more applications are no longer being released for 32-bit, even though they don't require more than 2GB of RAM to work.
> 
> I'm interested in hearing what obstacles there would be towards such a solution.
> 
> Thank you for your consideration.
> 
> Some other related projects that might be useful to mention here:
> 
> * [macosxbootloader](https://github.com/Piker-Alpha/macosxbootloader), boot loader for older 64-bit Macs with 32-bit EFI firmware, to be able to boot newer 64-bit versions of Mac OS X, using the same mixed mode (thunking) mechanism
> * A (currently still open) issue about properly supporting [(U)EFI Mixed Mode in systemd](https://github.com/systemd/systemd/issues/17056)

This is an issue tab for a reason. Not a fucking feature request. Are you people really this fucking stupid!?!?

volkertb, 2024-01-21:
There's no need to be so rude. :confused: And feature requests are also opened as issues. And then they can be assigned a separate `enhancement` label. But I don't have the rights to do that myself.

I've used GitHub enough over the years to understand how that works.

superluig164, 2024-01-21:
Hopefully this isn't too much of a necro, but I have been looking into ways to get Windows x64 booting on these tablets for quite a long time, and if this ever gets implemented or even partially implemented it might be the holy grail.  I'm not a developer unfortunately, but I'd be happy to help conduct testing or any kind of help I can give to make something like this work.  Unfortunately it seems like the issue is compounded by Ubuntu and ChromeOS supporting the setup natively already, leading to barely anyone seeing the point in putting in the work to hack around something like this.  Either way, I hope it goes somewhere, just writing to add my +1 I suppose.

volkertb, 2026-02-25:
One interesting project to watch in this regard is [CSMWrap](https://github.com/FlyGoat/CSMWrap), which implements Legacy BIOS compatibility for UEFI systems that lack a Compatibility Support Module (CSM).

CSMWrap has both ia32 and x64 builds. Through CSMWrap, it should, at least in theory, be possible to boot operating systems that require a 64-bit UEFI on a system that has only a 32-bit UEFI, even if it doesn't have a CSM. The boot chain would then be as follows:

32-bit UEFI ->CSMWrap with SeaBIOS implementation -> Clover 64-bit -> 64-bit OS

I haven't tried this yet, though. Anybody willing to experiment this, please keep us posted!

Also, please report any findings or feature requests for CSMWrap to https://github.com/FlyGoat/CSMWrap/issues.

lorn10, 2026-03-08:
Hi to all. 

I also found this super interesting topic via websearch and hopped for a more constructive end. Although the last post of @volkertb is of course absolutely helpful. Thanks for that! 👍 

For the sake of completeness it should be added here that there seems to exist (as of March 2026) already a way to boot into Linux Mint 22 (64-bit) without any additional boot-loader. This was at least confirmed at the following page regarding an **MacPro1,1** which used an original **Apple ATI Radeon HD 5870** card:

[AskDifferent: Mac Pro 1,1 Linux Boot Issue](https://apple.stackexchange.com/a/476797/390102)

The trick seems to be to set GRUB to the console mode by adding: `GRUB_TERMINAL_OUTPUT="console"

However, this most likely only works with **original Apple GPU hardware** which comes with native 32-bit EFI support in the vBios firmware. As far I know common PCIe UEFI GPU hardware usually only supports 64-bit UEFI. (In the past they had supported of course additionally also legacy BIOS mode.)

Finally, from a technical point of view I think this topic was started somehow at the wrong place. 😉 This should be effectively asked at the [rEFInd](https://www.rodsbooks.com/refind/) support page. Well, it was asked some years ago but the question remained [unanswered](https://sourceforge.net/p/refind/discussion/general/thread/e0b187ce95/). 

As far I know rEFInd always supported (and supports) 32-bit EFI Apple systems including a direct loading of 32-bit Linux kernels from the boot menu. So it might be possible to enhance rEFInd and add some sort of an EFI "mixed-mode" functionality which would also allow the loading of 64-bit Linux kernels:

32-bit UEFI -> rEFInd -> 64-bit Linux Kernel

I think that should make it possible to use also recent 64-bit UEFI based PCIe graphics cards in a MacPro1,1 (in conjunction with Linux). 🤔