CloverHackyColor/CloverBootloader #513: Can't update with any version after 5123, kernel panic

bigguygeo, 2022-04-27:
Laptops: Zbook 15 G3, 17 G4 crashing boot OS like High Sierra, Mojave, Catalina and others. 
Even if I do use Rehabman hot patch from scratch, it won't boot. Also tried to boot without any kexts except Virtual smc or fake smc, but no luck. Tried without acpi patched files... Also I have tested all variations of memory fixes like opencanopy or aptiofix

So laptop can boot only the versions are not higher than 5123, after it panic and this panic cant be captured, also boot log (-lastboot) not working.

kernel panic
https://ibb.co/sm5CJZ2

SergeySlice, 2022-04-27:
Nothing new.
Use release 5146. Compare config-sample.plist with your config.plist. Add Quirks section.
Replace **AptoFix***.efi to OpenRuntime.efi
Enjoy.

bigguygeo, 2022-04-27:
> Nothing new. Use release 5146. Compare config-sample.plist with your config.plist. Add Quirks section. Replace **AptoFix***.efi to OpenRuntime.efi Enjoy.

Wow! Thanks! It's working now! Boot is much faster!!! 

Do this quirks requires additional configuration, or that's enough to use settings from sample?

SergeySlice, 2022-04-27:
See WiKi about this.
Usually sample quirks are enough.

closed 2022-04-27 by SergeySlice