CloverHackyColor/CloverBootloader #518: Problems related to MacOS Ventura under Ivy Bridge systems

mjs520, 2022-06-07:
AvoidRuntimeDefrag is a must to boot Ivy Bridge systems. 
How to solve, do not close the AvoidRuntimeDefrag button, install MacOS Ventura, thank you.

SergeySlice, 2022-06-21:
Test please recent codes.

mjs520, 2022-06-24:
> > AvoidRuntimeDefrag is a must to boot Ivy Bridge systems.
> > How to solve, do not close the AvoidRuntimeDefrag button install MacOS Ventura, thank you.
> 
> Were you able to install it on Ivy Bridge? I have a 3930K, when I enter the system after all the installation steps, when the system is loading the computer is restarting. Not complete to enter in system in first time. Are you having the same problem?

Please refer to this article：https://www.insanelymac.com/forum/topic/351969-pre-release-macos-ventura/page/50/

SergeySlice, 2022-06-29:
Check release 5147.

SergeySlice, 2022-07-02:
It works.

closed 2022-07-02 by SergeySlice