CloverHackyColor/CloverBootloader #521: Ventura stuck at NVMe?

alessaba, 2022-06-24:
Hi,
I know this is not really a support page, but I tried to update Monterey to Ventura using the latest build, and the installation indeed went ahead. (before it would just reboot loop at the first update reboot)

The problem I am finding, though, is that at the third (I believe that's the final reboot, at the end of which you have access to the system) reboot, it got stuck, I think at the NVMe step of the boot. I remember reading somewhere that Ventura got a new NVMe driver or something like that, but I can't find that information anymore. Did somebody have the same problem or knows a solution, perhaps?

I am using the latest Lilu, Whatevergreen kexts, am on Coffee Lake (8700k)

SergeySlice, 2022-06-26:
I also encounter such panic and not resolve the origin of the problem.
But I resolved this by changing SMBIOS to model iMac17,1. The last installation stage was continued.
Try and report.

alessaba, 2022-06-26:
I can't get the disk to boot with that SMBIOS, it just reboots as soon as i press the boot entry
using iMac19,2 shows a slightly different result. I think the problem is related to the new "Cryptex" disk feature, and the panic when using iMac19,2 is caused by libsystemb.dylib, which libignition can't apparrently find. 

[image]

SergeySlice, 2022-06-29:
Check release 5147

alessaba, 2022-06-29:
The external SSD now boots, great!
I am going to try redoing the upgrade procedure on the internal NVMe, since there it got stuck but not panicked

closed 2022-06-29 by alessaba