CloverHackyColor/CloverBootloader #526: Clover 5147, 5148 NO Ventura Beta 3 update

HandaTakuya, 2022-07-10:
I currently in use macOS Ventura Dev Beta 2 and I check for update Dev Beta 3 ... BUT no update.
If it is Open Core : check Secure Boot Model
For Clover ... How to fix ?

[image]

SergeySlice, 2022-07-18:
What is your SMBIOS model?

HandaTakuya, 2022-07-21:
> What is your SMBIOS model?

MacBook Pro 16,1 and I Try iMac17,1 but not boot

jief666, 2024-01-04:
Do you still need this issue ?

dateno1, 2024-01-04:
No update on Ventura is not clover's problem (OC is same result)

It bug on MacOSX

Some of Intel Mac has problem

jief666, 2024-01-05:
Ok. Closing.

closed 2024-01-05 by jief666

dateno1, 2024-01-06:
Add RestrictEvents.kext and Boot-Args - revpatch=sbvmm

Plese test it (It maybe fix problem)