CloverHackyColor/CloverBootloader #528: Can't load macOS partition to boot if /EFI/CLOVER/drivers/UEFI/ApfsDriverLoader.efi

oldman20, 2022-07-15:
Hello, anytime I have loadapfsdriver.efi driver in the clover driver folder i get a black screen when booting to clover. I am able to get around this by moving the driver outside of the driver folder. That gets me to the clover screen at which point i can enter the efi shell. From there i can manually load the loadapfsdriver.efi. Once i exit the shell I can see the apfs partitions and boot as normal. Is there anyway i can fix it or automate this process? 

This is my specs
Dell Inspiron 3558 (UEFI) - Intel Core i5 (5th Gen) 5200U /2.20GHz - 8GB RAM DDR3L - Intel(R) HD Graphics 5500/ nVIDIA GeForce 920M - 1366x768 - Realtek ALC255 @ Intel Wildcat Point-LP PCH - High Definition Audio Controller [F0] - Intel(R) Dual Band Wireless-AC 3160 - Dell Touchpad /Synaptics - Samsung SSD 860 EVO 250GB (GPT)

SergeySlice, 2022-07-15:
I was not standing behind you but I know you do something wrong.
ApfsDriverLoader.efi works fine in thousands computers.

oldman20, 2022-07-19:
> 

i do same Dell 3558 like others computer i touch, and i think smt with 3558 not like others

SergeySlice, 2022-07-21:
Try replace it by other driver:
[AppleAPFS.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9161913/AppleAPFS.efi.zip)

oldman20, 2022-07-24:
> Try replace it by other driver:
> [AppleAPFS.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9161913/AppleAPFS.efi.zip)
I tried but just blackscreen, no load GUI Clover. it can boot if load manual driver in efi shell
[2022-07-24_01-42_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/9174963/2022-07-24_01-42_BOOTX64.EFI.log)

oldman20, 2022-07-24:
Here Prelboot.log when boot from Clover until show GUI then press F2:
`0:100  0:100  MemLog inited, TSC freq: 2194917425
0:100  0:000  CPU was calibrated with ACPI PM Timer
0:100  0:000  CloverX64 : Image base = 0x8D1DA000
0:100  0:000  Clover ImageHandle = 9fe22e98
0:104  0:004  === [ Starting Clover ] =========================
0:104  0:000  Now is 22.07.2022,  10:36:08 (GMT)
0:104  0:000  Starting Clover revision: 5148 (master, commit 8920835bf) on American Megatrends EFI
0:104  0:000  Build id: 20220709065223-8920835-5148-dirty
0:104  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG -t GCC53 | --cmd-len=50000 -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG --conf=/Users/sergey/src/CloverBootloader/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 13]
0:104  0:000  === [ Get Smbios ] ==============================
0:104  0:000  Type 16 Index = 0
0:104  0:000  Total Memory Slots Count = 2
0:104  0:000  Type 17 Index = 0
0:104  0:000  SmbiosTable.Type17->SlotIndex = 0
0:104  0:000  SmbiosTable.Type17->ModuleSize = 8192MB
0:104  0:000  SmbiosTable.Type17->Type = 0
0:104  0:000  SmbiosTable.Type17->Frequency = 1600MHz
0:104  0:000  SmbiosTable.Type17->Bank/Device =  DIMM A
0:104  0:000  SmbiosTable.Type17->Vendor = Samsung
0:104  0:000  SmbiosTable.Type17->SerialNo = 96AD9C99
0:104  0:000  SmbiosTable.Type17->PartNo = M471B1G73DB0-YK0
0:104  0:000  Type 17 Index = 1
0:104  0:000  Ignoring insane frequency value 0MHz
0:104  0:000  Boot status=0
0:104  0:000  Running on: 'Inspiron 3558' with board '0G1TDD'
0:104  0:000  === [ GetCPUProperties ] ========================
0:104  0:000  CPU Vendor = 756E6547 Model=306D4
0:104  0:000   The CPU supported SSE4.1
0:104  0:000   The CPU supported RDRAND
0:104  0:000  BrandString = Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz
0:104  0:000   The CPU supported turbo
0:104  0:000   Energy PerfBias is  visible:
0:104  0:000  MSR 0x35               20004
0:104  0:000  MSR 0xE2 before patch 1E008404
0:104  0:000  MSR 0xE2 is locked, PM patches will be turned on
0:104  0:000  MSR 0xCE              0005053B_F3011600
0:104  0:000  corrected FLEX_RATIO = 0
0:104  0:000  MSR 0x1B0             00000000
0:104  0:000  FSBFrequency = 100 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
0:104  0:000  MaxDiv/MinDiv: 22.0/5
0:104  0:000  Turbo: 25/25/25/27
0:104  0:000  Features: 0x77FAFBBFBFEBFBFF
0:104  0:000  Threads: 4
0:104  0:000  Cores: 2
0:104  0:000  FSB: 100 MHz
0:104  0:000  CPU: 2200 MHz
0:104  0:000  TSC: 2200 MHz
0:104  0:000  PIS: 100 MHz
0:104  0:000  ExternalClock: 25 MHz
0:104  0:000  === [ GetDevices ] ==============================
0:104  0:000  GOP found at: PciRoot(0x0)/Pci(0x2,0x0)/AcpiAdr(0x80010100)
0:104  0:000  PCI (00|00:00.00) : 8086 1604 class=060000
0:104  0:000  PCI (00|00:02.00) : 8086 1616 class=030000
0:104  0:000   - GFX: Model=Intel HD Graphics 5500 (Intel)
0:104  0:000   - GOP: Provided by device
0:104  0:000  PCI (00|00:03.00) : 8086 160C class=040300
0:104  0:000  PCI (00|00:14.00) : 8086 9CB1 class=0C0330
0:104  0:000  PCI (00|00:16.00) : 8086 9CBA class=078000
0:104  0:000  PCI (00|00:16.01) : FFFF FFFF class=FFFFFF
0:104  0:000  PCI (00|00:1B.00) : 8086 9CA0 class=040300
0:104  0:000  PCI (00|00:1C.00) : 8086 9C90 class=060400
0:104  0:000  PCI (00|00:1C.02) : 8086 9C94 class=060400
0:104  0:000  PCI (00|06:00.00) : 8086 08B3 class=028000
0:104  0:000   - WIFI: Vendor= Intel
0:104  0:000  PCI (00|00:1C.04) : 8086 9C98 class=060400
0:104  0:000  PCI (00|08:00.00) : 10DE 1299 class=030200
0:104  0:000  PCI (00|00:1D.00) : 8086 9CA6 class=0C0320
0:104  0:000  PCI (00|00:1F.00) : 8086 9CC3 class=060100
0:105  0:000  PCI (00|00:1F.02) : 8086 9C83 class=010601
[... 1824 lines cut ...]
0:118  0:000  '\EFI\CLOVER\smbios.plist' not loaded. Efi error Not Found
0:118  0:000  Using latest BiosVersion from config instead of default 'MBP121.88Z.F000.B00.2106131845'
0:955  0:836  Using latest BiosReleaseDate from config instead of default '06/13/2021'
0:981  0:025  BiosVersion: MBP121.88Z.0176.B00.1804091715
1:006  0:025  BiosReleaseDate: 04/09/2018
1:032  0:025  EfiVersion: 427.140.8.0.0
1:058  0:025  Calibrated TSC Frequency = 2194918185 = 2194MHz
1:083  0:025  === [ GetListOfThemes ] =========================
1:109  0:025  No theme dir was discovered
1:135  0:025  === [ Found config plists ] =====================
1:160  0:025  - config.plist
1:190  0:030  === [ LoadDrivers ] =============================
1:219  0:029  Loading AudioDxe.efi status=Success
1:304  0:084   - driver needs connecting
1:330  0:025  Loading AppleAPFS.efi status=Success
1:896  0:565   - driver needs connecting
1:921  0:025  2 drivers needs connecting ...
1:947  0:025  PlatformDriverOverrideProtocol->GetDriver overriden
1:973  0:025  APFS driver loaded`
 
 
AfterLoadAPFSManual_F3
[image]

[image]

oldman20, 2022-07-24:
[image]

SergeySlice, 2022-07-27:
> Try replace it by other driver:
> [AppleAPFS.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9161913/AppleAPFS.efi.zip)
> I tried but just blackscreen, no load GUI Clover. it can boot if load manual ApfsDriverLoader.efi driver in efi shell
> [2022-07-24_01-42_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/9174963/2022-07-24_01-42_BOOTX64.EFI.log)

Your log ended here
`1:957  0:000  APFS driver loaded
1:957  0:000  Searching for invalid DiskIo BY_DRIVER connects: not found, all ok
4:118  2:161  === [ InitScreen ] ==============================

`

So your drive somehow has invalid DiskIo. Sata sriver can't read it. May be there is a BIOS protection for the sectors.

oldman20, 2022-07-28:
> > Try replace it by other driver:
> > [AppleAPFS.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9161913/AppleAPFS.efi.zip)
> > I tried but just blackscreen, no load GUI Clover. it can boot if load manual ApfsDriverLoader.efi driver in efi shell
> > [2022-07-24_01-42_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/9174963/2022-07-24_01-42_BOOTX64.EFI.log)
> 
> Your log ended here
> 
> ```
> 1:957  0:000  APFS driver loaded
> 1:957  0:000  Searching for invalid DiskIo BY_DRIVER connects: not found, all ok
> 4:118  2:161  === [ InitScreen ] ==============================
> ```
> 
> So your drive somehow has invalid DiskIo. Sata driver can't read it. May be there is a BIOS protection for the sectors.

Thank, At least there's a clue! Is it possible solve or bypass?

oldman20, 2022-07-28:
Forgot, i use this specs:
Dell Inspiron 3558 (UEFI) - Intel Core i5 (5th Gen) 5200U /2.20GHz - 8GB RAM DDR3L - Intel(R) HD Graphics 5500/ nVIDIA GeForce 920M - 1366x768 - Realtek ALC255 @ Intel Wildcat Point-LP PCH - High Definition Audio Controller [F0] - Intel(R) Dual Band Wireless-AC 3160 - Dell Touchpad /Synaptics - Samsung SSD 860 EVO 250GB (GPT)
And replaced CDRW slot with HDD 640 CaddyBay

SergeySlice, 2022-08-02:
Some debugging on your hardware needed. I have no access to your hardware and can't reproduce the bug on other hardware.

oldman20, 2022-08-02:
so sad, clover still working with boot manager for windows, linux. Stranger!

oldman20, 2022-08-11:
> Some debugging on your hardware needed. I have no access to your hardware and can't reproduce the bug on other hardware.

the funny i found this time is when test with Clover 4045, i put ApfsDriverLoader.efi from 5148 to EFI driver folder of 4045, APFS partition is showed 
i attached preboot.log from 4045 clover, if you want to check, thank
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/9307603/preboot.log)
[image]
!

SergeySlice, 2022-08-15:
Nobody still reproduce your bug.

oldman20, 2022-08-15:
so maybe is this a Exception about hardware?

oldman20, 2022-08-23:
Finally i can control this issue, can sure reason cause my "special" behavior while load efi drivers. 

All what i did is set ApfsDriverLoader.efi (and HfsPlus.efi, if need for installation macOS with Clover, yes)

Firstly put ApfsDriverLoader.efi and HfsPlus.efi in root of EFI partition. 

Boot into Clover GUI, press S to open EFI Shell:

fsx: #x depend your partitions layout, mine is 0
ls #check for sure x is correct part, if that you will see ApfsDriverLoader.efi and HfsPlus.efi
bcfg driver add y fsx:\ApfsDriverLoader.efi "What ever name"
In my case, I add both ApfsDriverLoader.efi and HfsPlus.efi so these my cmd:

bcfg driver add 0 fs0:\ApfsDriverLoader.efi "Apfs Driver"

bcfg driver add 1 fs0:\HfsPlus.efi "HFS+ Driver" 

Then type: exit to out of EFI She'll/ or just Ctrl+Alt+Del to reboot device, Clover will autoload EFI and show macOS to boot

*NOTE: bcfg will store EFI drivers in NVRAM, so if reset NVRAM will be gone, must repeat above process

Hoping Clover new version "fix" for special device like this:
[image]

SergeySlice, 2023-01-05:
Resolved

closed 2023-01-05 by SergeySlice

oldman20, 2023-01-05:
> Resolved

yes, resolved temporary :(