CloverHackyColor/CloverBootloader #530: WIKI: Tables are not formatted correctly

5T33Z0, 2022-07-28:
**Examples**:

https://github.com/CloverHackyColor/CloverBootloader/wiki/Configuration#ig-platform-id
https://github.com/CloverHackyColor/CloverBootloader/wiki/Configuration#smartups
https://github.com/CloverHackyColor/CloverBootloader/wiki/Configuration#fbname
https://github.com/CloverHackyColor/CloverBootloader/wiki/Configuration#chassistype

SergeySlice, 2022-07-29:
Mostly fixed.
Look please any else formatting mistakes.

5T33Z0, 2022-07-29:
All fixed. 

A small suggestion:  I wouldn't put tables in fenced code, otherwise they are not formatted correctly. Instead, I would format them like this:

```
Value |  Type  | Power Supply Type
------|------- | ----------------- 
1   |  Desktop  | AC 
2  |  Mobile   | Battery 
3  | Server    | SmartUPS
```

Which results in a nice Table:

Value |  Type  | Power Supply Type
------|------- | ----------------- 
1  |  Desktop  | AC 
2  |   Mobile   | Battery 
3  | Server   | SmartUPS

closed 2022-07-29 by 5T33Z0