CloverHackyColor/CloverBootloader #536: Can't boot on disk connected to usb3 extension card, XhciDxfe.efi driver not working

vinceFR7528, 2022-08-08:
Hi,

My old GA-E7AUM-DS2H BIOS mobo has only USB2 and when it boots on a USB live disk it uses the fullspeed 12Mb (not Hi-Speed, strange?) so booting a linux ISO takes a long long time. I added a PCIe USB3 extension card, this card is seen by the bios but not used so the disk connected to it is not seen from the boot loader. I installed CLOVER 5148 and tried to load the XhciDxfe.efi driver, but it doesn't work as CLOVER doesn't detect the usb disk.

My PCIe USB3 extension card is compliant with USB standard, so no specific driver is needed, chipset Renesas µPD720201.

SergeySlice, 2022-08-09:
The question requires debugging which can be provide by the controller owner.

vinceFR7528, 2022-08-09:
ok, how can I do debbugging?

SergeySlice, 2022-08-15:
Compile your own version of XhciDxe with additional DEBUG() at every line you want to see. 
Then dlete these lines and add new lines. Try again.
Again and again until the problem will be localized.

SergeySlice, 2023-01-05:
See Release 5151, may be it helps.

SergeySlice, 2023-09-05:
abandoned

closed 2023-09-05 by SergeySlice