CloverHackyColor/CloverBootloader #537: In does not support N card installed

jackacc, 2022-08-09:
Clover patch installation N CARDS, the SIP testing banned AMFI, disable Metal and open N CARDS the Arguments added
		<key>CsrActiveConfig</key>
		<string>0xFEF</string>
		<key>Arguments</key>		
<string>amfi_get_out_of_my_way=0x1 ngfxcompat=1 ngfxgl=1 nvda_drv_vrl=1</string>

There is no effect

Opencore boot - args added 
csr-active-config FF0F0000
                    <key>boot-args</key>
                    <string>amfi_get_out_of_my_way=0x1 ngfxcompat=1 ngfxgl=1 nvda_drv_vrl=1</string>
successful

SergeySlice, 2022-08-10:
Give me a link what are you trying to do. What is "N CARDS"?

jackacc, 2022-08-15:
> 给我一个链接你想做什么。什么是“N卡”？

NVIDIA GeForce GT220M， 320M， 8500GT，8600GT，9500GT，9600GT，GT210,GT220，GT240 
Geforce-Kepler-patcher 与Opencore-Legacy-Patcher打补丁

SergeySlice, 2022-08-18:
https://github.com/chris1111/Geforce-Kepler-patcher

SergeySlice, 2022-08-18:
It is for Clover.

SergeySlice, 2022-08-18:
> > 给我一个链接你想做什么。什么是“N卡”？
Is it Chinese? I can't even translate.

jackacc, 2022-08-21:
> > > 什么是“N卡”？
> > > 是中文的吗？我连翻译都不会。

你会中文，我就不用工具翻译英文了
[image]

禁用AMFI在clover怎么做呢？上面图是opencore设置
[image]
不禁用AMFI上图按钮灰色无法安装显卡驱动补丁

SergeySlice, 2022-08-21:
Again, do not write in chinese.

closed 2022-08-28 by SergeySlice

jackacc, 2022-09-04:
> Again, do not write in chinese.

My notebook is a generation of motherboard without nvram hardware, a BBS someone reply me using nvram. The plist to launch parameters inside, he said that the Arguments does not support long code amfi_get_out_of_my_way = 0 x1 beyond number