CloverHackyColor/CloverBootloader #538: N clover - Patcher gray can't patch card driver

jackacc, 2022-08-17:
[image]

启动参数中打入amfi_get_out_of_my_way=0x1  ngfxcompat=1 ngfxgl=1 nvda_drv_vrl=1四条，他们分别是禁用AMFI、禁用Metal和开启N卡检测，还有就是需要在csr-active-config中写入FF0F0000来禁用SIP

OpenCore- Misc-Security（安全）-SecureBootModel:Disabled
[image]

clover?

SergeySlice, 2022-08-18:
I can't read chinese. Make the screenshot with english interface,

jackacc, 2022-08-21:
> 我看不懂中文。用英文界面制作截图，
OpenCore- Misc-Security-SecureBootModel:Disabled
[image]
[image]

Where is the clover options?

SergeySlice, 2022-08-28:
The secure boot model in Clover is always "disable".

closed 2022-09-04 by SergeySlice

jackacc, 2022-09-04:
> The secure boot model in Clover is always "disable".
My notebook is a generation of motherboard without nvram hardware, a BBS someone reply me using nvram. The plist to launch parameters inside, he said that the Arguments does not support long code amfi_get_out_of_my_way = 0 x1 beyond number