CloverHackyColor/CloverBootloader #541: Build Error: fatal: No names found, cannot describe anything.

stevezhengshiqi, 2022-08-19:
Hi, thank you guys for maintaining this boot loader.

I see CI is reporting errors for recent commits, and I am trying to fix it. However, as I build the latest commit https://github.com/CloverHackyColor/CloverBootloader/commit/806c09e3f06c58567bd53c8f19f08adaf6f20837 on my laptop with Monterey, it reports error:
```log
Running edk2 build for CloverX64 using the command:
build --cmd-len=50000 -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG --conf=/Users/steve/CloverBootloader/Conf -D USE_LOW_EBDA -p Clover.dsc  -a X64 -b RELEASE -t GCC53 -n 9 

fatal: No names found, cannot describe anything.
fatal: No names found, cannot describe anything.
fatal: ambiguous argument 'tags/..HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
```

And it reports the same error on my local repository with CI.
My build command is simple: `./buildme "" ci `

SergeySlice, 2022-08-23:
What is CI?

SergeySlice, 2022-08-23:
- Done -
Build end time: 20:03:40, Aug.23 2022
Build total time: 00:03:55

  -> CloverX64-RELEASE_GCC53-20220818195924-806c09e-5149.efi

SergeySlice, 2022-09-04:
Seems resolved

closed 2022-09-04 by SergeySlice