CloverHackyColor/CloverBootloader #542: PCI/NVMe Boot Drive Not Shown on New Server and Not Autobooting

evoelise, 2022-08-21:
I had Clover boot set up on a USB on a Dell 720xd to boot from a NVMe drive on an internal PCIe slot. This had all worked fine. 

I now have taken the USB and the same NVMe drive and PCIe card and place these on a different but similarily specced Dell 720xd.

Clover loads but does not show any disks to boot from by default. If I press F3 the NVMe/PCI drive shows up (with a werid name with all _____ underscores) and I can manually boot from that OK. 

But why does this drive not show up as normal?

Also I have DefaultVolume set to "LastBootedVolume" in the config.plist but it does not ever autoboot?

What is going wrong here? How can I diagnose the issue?

I have tried creating a new Clover boot USB but this has the same problem.

Morbius01, 2022-08-22:
The same question was discussed in another forum. From this discussion, it came out that Clover cannot boot from your NVMe drive because the OS was installed on a different system.

I have myself tried to get Clover to boot from a previously installed OS and it never worked.

You may like to try to re-install your OS on the NVMe drive in the new system using the newly created Clover key and see if it works.

evoelise, 2022-08-22:
> it came out that Clover cannot boot from your NVMe drive because the OS was installed on a different system.

Not sure if this is the case here. The OS was installed fresh on this server - although the Clover boot install on the USB was the same. 

Although I also tried a brand new Clover install on a different USB and got the same issue. 

I will try a brand new Clover and do a brand new OS install on the server too and see how that works.

Morbius01, 2022-08-22:
My experience with Clover is that it can be quite tricky to install. But, when it’s correctly installed, it works wonderfully well. See my post  [here](https://github.com/CloverHackyColor/CloverBootloader/discussions/491).

If you succeeded on your first system, you may like to replicate exactly the same process from scratch on your second system, meaning you create a new Clover key and you install your OS again on the NVMe. If the same method does not work, you may like to consider debugging your second system.

As for my own experience, the only way I succeeded in installing an OS on a non-bootable drive, which I can boot with Clover, was to create a Clover key in a first step, and then, to install the OS **from this Clover key** in a second step. This is how I managed to install Windows 11 on a 2009 (motherboard / BIOS / CPU) system !  However, because my system is quite old (non UEFI, legacy BIOS), things might be different on a more recent platform.

You might consider using this method if anything else fails for you.

Which OS are you trying to install on your NVMe disk ?
Which version of Clover are you using ?

Let us know how things work for you...

evoelise, 2022-08-22:
You may like to try to reinstalled the OS on the NVMe drive from scratch then used a formatted USB with a new install of Clover on it.

So replicated what worked the first time I did this. 

Same issue - Clover does not show the installed OS/NVMe by default. If I press F3 it is shown and I can manually boot from this.

So Clover "knows" about this OS boot disk - it just hides it for some reason and will not boot from it automatically.

Why does this drive not show up as normal?

Morbius01, 2022-08-22:
> Why does this drive not show up as normal?
If I was you, I would not focus on trying to get an answer to this question because it might make you waste a lot of time...

What I would do is to follow the method which I mentioned in my previous post and which has always worked for me : install the OS from scratch on your NVMe **_after booting from the Clover key_** (and not independently from Clover).

By the way, you have not answered any of the 2 questions I asked in my previous post. Those points are important because the methods I used to install Windows and Linux are not the same.

evoelise, 2022-08-22:
Clover version is the latest 5149.
OS is TrueNAS Core v13

Can you explain exactly what you mean by "install the OS from scratch on your NVMe after booting from the Clover key (and not independently from Clover)." please? Do you mean boot from a new Clover install and then use that to boot the installer (with is on another USB) and install on NVMe?

Morbius01, 2022-08-22:
> .....Do you mean boot from a new Clover install and then use that to boot the installer (with is on another USB) and install on NVMe?

Exactly.
First, you create a new Clover key. Then you boot from it. When the Clover menu gets displayed, you insert the other USB drive with the OS installer on it. Then, you select the new entry which appears in Clover menu. This should normally starts the OS installation process, which you need to follow. You should keep the Clover key in until the end of the entire installation process.

evoelise, 2022-08-22:
Thanks for the response - I did all that and still have the exact same issue - the NVMe drive only shows when I press F3 and so does not auto boot. 

However I checked back through my notes and realised that when I did this before I was trying out TrueNAS Scale and not TrueNAS Core. TrueNAS Scale is Linux based whilst TrueNAS Core is BSD Unix based. I guess Clover doesn't work properly with BSD.

Doing some more google I found various pages with problems with Clover and BSD and they recommend using rEFInd bootloader instead. I tried rEFInd and it worked first time.

So this issue can be closed but thanks for your assistance anyway.

closed 2022-08-22 by evoelise

evoelise, 2022-08-22:
https://www.truenas.com/truenas-core/