CloverHackyColor/CloverBootloader #543: Updating Clover 5142, no bootable drive

guidocioni, 2022-08-25:
I've been trying to update Clover 5142 to something new.
I've tried both 5146 and 5149 but every time I try to update the disk does not boot anymore and I see the message "Insert bootable drive" after the BIOS screen; I don't even get to the Clover screen. 

What kind of component was changed from 5142 that could cause this? Or is the BIOS version too old? 

I have a Gigabyte Z370M D3H with an i7-8700 running MacOs Catalina. I'm not planning on running Ventura but I may update to Monterey in the future (still not sure about it as Catalina is working just fine).

allegrox, 2022-08-27:
The problem I encountered is similar to yours. Since the clover 5146, the installer cannot update the files of the EFI partition, but it will delete the UEFI boot entry, so I only manually replace the files to update.

guidocioni, 2022-08-27:
> The problem I encountered is similar to yours. Since the clover 5146, the installer cannot update the files of the EFI partition, but it will delete the UEFI boot entry and CLOVERX64.efi, so I only manually replace the files to update.

Thanks, so you basically did the update and then used these two files from the old installation?

SergeySlice, 2022-08-28:
Two files? Why I will use files from old installation?

guidocioni, 2022-08-31:
> Two files? Why I will use files from old installation?

I'm just trying to understand how can I update the clover bootloader while still having a bootable system at the end 😉

allegrox, 2022-09-03:
I compiles clover at my pc, and then overwrites several files such as CLOVERX64.efi and ApfsDriverLoader.efi in the compilation result into the EFI partition to complete the update.

allegrox, 2022-09-03:
> > Two files? Why I will use files from old installation?
> 
> I'm just trying to understand how can I update the clover bootloader while still having a bootable system at the end 😉
> 
> @allegrox could you please specify exactly the steps that you had to take to update clover but still being able to boot? thanks

I compiles clover at my pc, and then overwrites several files such as CLOVERX64.efi and ApfsDriverLoader.efi in the compilation result into the EFI partition to complete the update.

SergeySlice, 2022-11-02:
It must work

SergeySlice, 2023-01-05:
I see no Clover issue.

closed 2023-01-05 by SergeySlice