CloverHackyColor/CloverBootloader #545: Current Beta Releases break Ventura boot on iMac20,x

5T33Z0, 2022-09-05:
I just wanted to inform you that the last 2 beta releases of Clover (after 6890d4812) crash  when trying to boot macOS Ventura. SMBIOS iMac20.2

I am reverting back to 6890d4812 for the time being

5T33Z0, 2022-09-06:
Clover beta-9cdae6f9d  Breaks Ventura ad Monterey as well.

5T33Z0, 2022-09-07:
Builds past (after https://github.com/CloverHackyColor/CloverBootloader/commit/6890d4812413d2e199fef3b058ce93e0f348707c) also Crash Monterey on MacBookPro10,1.

taruyato, 2022-09-09:
confirmed on my laptop with SMBIOS MBP15,2, start from https://github.com/CloverHackyColor/CloverBootloader/commit/6890d4812413d2e199fef3b058ce93e0f348707c to lastest commit, Clover crash when trying to boot to Ventura.