CloverHackyColor/CloverBootloader #550: Clover boot problem

StalkerJS, 2022-09-22:
Hi.
Have problem with clover booting on two computers last few days. I use it for boot windows from pci-e ssd on old pc.
BDU make error "not found standart clover archive". I load 5149 iso from there and give it to BDU. Is make flash-drive w\out errors after it. But I see this after load on PC: 
[image]

What should I do for fix this? thx

SergeySlice, 2022-09-26:
Make a copy \EFI\BOOT\BOOTX64.EFI to \EFI\CLOVER\CLOVERX64.EFI.
It is the same fime with different names.

jackacc, 2022-09-27:
Can have the old computer lack BIOS_BlockIO, do not use change BOOT6 BOOT7, BOOT7 renamed the BOOT, with EFI directory

4C3T, 2022-09-28:
> Some old computers lack BIOS_BlockIO, don't use BOOT6, into BOOT7, BOOT7 renamed the BOOT, replace the original instead [boot.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9657191/boot.zip)

I'm also having this problem, and this description of yours didn't make much sense.
Are we supposed to rename these or what exactly are you trying to explain here?

I get "Some old computer lack BIOS_BlockIO, don't use BOOT6" then it's "Que?/Huh?/Waaat?" for me and probably everyone else trying to decipher that...

I did however try to rename boot to boot.bak and replaced it with the zipped one you linked to, but all I got was a lot of red Exception Type 0000000000000000D (<--Give or take a few 0), so something tells me this file was wrong.
I tried it on 2 different computers, one an old Lenovo Core2 Duo laptop and an ASUS P8H77-V_LE mobo with an i5 @3.4GHz, both yielded the same response.

jackacc, 2022-09-28:
> I'm also having this problem, and this description of yours didn't make much sense. Are we supposed to rename these or what exactly are you trying to explain here?
>
> I get "Some old computer lack [...]

HFSPlus. The problem of efi, in one, the author extracted HFSPlus. Efi, I provide the author original efi
[EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9660637/EFI.zip)

About the boot，Clover author said as early as 10 years ago:

[Clover](https://applelife.ru/tags/Clover/) на NForce MCP79
0) Параметры BIOS: SATA=AHCI(Asus K50Id) или SATA=IDE
1) Установить [Clover](https://applelife.ru/tags/clover/) r798 (или более новей) на USBFlash(FAT32) (можно использовать CardReader у кого есть)
2) скопировать ION-64.efi в /EFI/drivers64/ION-64.efi

ION-64.efi   Is the boot7 now
[image]
[image]

[
[image]
boot.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9660691/boot.zip)

closed 2022-10-27 by SergeySlice