CloverHackyColor/CloverBootloader #551: Ventura avx2

jackacc, 2022-09-27:
With macOS Ventura, Apple finally dropped the last Mac that lacked the [AVX2.0 CPU instruction], the 2013 Trash Can Mac Pro (MacPro6,1). With this, systems lacking AVX2.0 can no longer boot Ventura natively as Apple has stripped the legacy non-AVX2.0 dyld shared caches from the OS. However due to compatibility issues with Rosetta 2, Apple is forced to retain a pre-AVX2.0 dyld shared cache on Apple Silicon systems.

Thus to support older machines, this kext will force the macOS installer/updater to install the Apple Silicon Cryptex (OS.dmg) instead of the stock Intel variant. More information can be found under

For this kext to work, you'll also need the following Kernel->Patch installed in OpenCore:

Arch: x86_64
Base: _authenticate_root_hash
Count: 0
Find:
Identifier: com.apple.filesystems.apfs
Limit: 0
Mask:
MaxKernel:
MinKernel: 22.0.0
Replace: B800000000C3
ReplaceMask:
Skip: 0

Ommission of this kernel patch will result in APFS.kext Kernel Panicing due to Cryptex hash not matching expected hash.

[CSLVFixup.kext.zip](https://github.com/CloverHackyColor/CloverBootloader/files/9657263/CSLVFixup.kext.zip)

jackacc, 2022-09-27:
Clover > config. How will the plist OC code conversion to come over

SergeySlice, 2022-09-28:
KextToPatch
Count = 1
Find         = <000000000000>
MaskFind = <000000000000>
Replace    = <B800000000C3>
Name = com.apple.filesystems.apfs
Procedure = _authenticate_root_hash
MatchOS = 13.0
Skip = 0

closed 2022-10-27 by SergeySlice