CloverHackyColor/CloverBootloader #559: Can't boot in virtualbox after 5108

shilkazx, 2022-10-01:
I use clover iso as a optical boot device for a long run. But after release 5115(next normal release after 5108), clover iso malfunction just as the screenshot attached.
[image]

OS: Archlinux x86_64
VirtualBox version: 6.1

jackacc, 2022-10-04:
VMware Workstation Pro

abaumgaertner, 2022-11-09:
Tested on Windows 10 22H2 x64, VirtualBox 6.1.40, Clover-5150

Clover GUI starts normally.

Check your settings for VM:
General / Type can be set to any OS BUT needs to be 64bit variant.

eg.

[image]

closed 2023-01-05 by SergeySlice