CloverHackyColor/CloverBootloader #563: Clover can't see PCIE M.2 SSD on ProLiant DL 580 G4

lovemini, 2022-10-09:
I boot the computer in legacy BIOS mode, because the machine does not support EFI.  I use boot7 to boot Clover (boot6 dont't work).  Clover can't see my  PCIE M.2 SSD.  I am sure NvmExpressDxe.efi was installed.  Hardware is no problem because windows can see the SSD. 
Please help me  to solve this problem?

[image]
[image]
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/9741875/preboot.log)
[drivers.txt](https://github.com/CloverHackyColor/CloverBootloader/files/9741876/drivers.txt)
[map.txt](https://github.com/CloverHackyColor/CloverBootloader/files/9741877/map.txt)

Morbius01, 2022-10-10:
If I am correct, HP ProLiant DL 580 G4 is a system which dates back from years 2010-2012.

Could you let us know exactly what makes you think that Clover cannot see your PCIE M.2 SSD ?

Could you also let us know :
- What version of Clover you use ?
- What version of Windows you wish to install ?

Most importantly, have you started to start the installation of your Windows on your  PCIE M.2 SSD through Clover ?

lovemini, 2022-10-11:
The Clover version is 5149(lastest).

I installed windows 11 enterprise in UEFI mode using Clover(with NvmExpressDxe.efi).

Windows can only be installed on HDD disk because Clover can't see my PCIE M.2 SSD.

lovemini, 2022-10-12:
After a long time testing, i find the reason of the problem. It is caused by boot6 and boot7.
Boot6 has its own device driver, and boot7 build with BiosBlockIO driver.

The ProLiant DL 580 G4 has RAID controller (Smart Array P400), all the HDD disk was attached on the RAID Card. Clover file was placed in HDD (not USB Key).

I make a Clover USB boot disk using BDUtility. If boot with USB key and using boot6 to boot Clover, it can see USB Key and NVME SSD(can boot from NVME SSD), but can't see HDD disk.

MBR->boot0ss->boot6(on HDD)->can't see RAID HDD, boot failed.
MBR->boot0ss->boot7(on HDD)->can see RAID HDD and boot into Clover GUI, but can't see NVME SSD (can't boot form NVME).

I think boot6 do scan the hardware and drive it, but boot7 don't.

So i have to use a USB Key to boot the computer.  
I want to boot from RAID HDD, not USB Key, who can help me to solve the problem?

Morbius01, 2022-10-12:
If you still want to install Windows 11 on your PCIe SSD, don't worry about boot6 or boo7. Just follow the instructions I described in my previous post. It works (I did it) and it's the only way.

lovemini, 2022-10-13:
Thanks Morbius01.  I have install Windows 11 on PCIE SSD boot from  USB Key with Clover 5149 installed. 
All is ok.

Morbius01, 2022-10-13:
Great !  Congrats !
I am happy I could help you.

closed 2022-11-06 by SergeySlice