CloverHackyColor/CloverBootloader #565: Cannot automatically boot from hidden volumes

lostboy7, 2022-10-18:
A volume detected as hidden, for example as shown in this log excerpt

7:450  0:013  - [05]: 'EFI'
7:479  0:028        AddLoaderEntry for Volume Name=EFI, idx=1
7:493  0:013        OSVersion= 
7:506  0:013        IconName=unknown comma=-1 size=7
7:520  0:013        Full=unknown
7:535  0:015        Loader entry created for 'PciRoot(0x0)\Pci(0x3,0x2)\Pci(0x0,0x0)\NVMe(0x1,97-BD-B4-81-55-38-25-00)\HD(1,GPT,CFA63771-8686-460F-96A5-59AB2D417C5A,0x40,0x32000)\EFI\BOOT\BOOTX64.efi'
7:549  0:013        Menu entry added at index 1
7:563  0:013       hiding entry because DiskKind is DISK_KIND_INTERNAL: \EFI\BOOT\BOOTX64.efi   <------

Cannot be automatically booted from.   Seems the  "<key>DefaultVolume</key>" config.plist item ignores any volumes that are so marked.

Related - there appears to be no way to control this behavior either of marking a volume hidden (I know you can mark additional volumes as hidden, but the ones Clover decides to mark cannot be "unmarked").    And, there seems to be no way to change the default to show all hidden volumes.   Any of these would be suitable workarounds. 

As it is, apparently the only way to boot such a detected hidden volume is to manually press F3 to show it, and then pick it.  This works fine.

SergeySlice, 2022-11-06:
What do you want to boot? BOOTX64.EFI? And what is Clover?
DISK_KIND_INTERNAL is usual HDD. It is always bootable, ab ovo.

lostboy7, 2022-11-07:
Thanks for the reply.

What am I trying to boot: I a, trying to boot a Linux installation.  Rocky 8.5 from a NVMe device to be specific.  For whatever reason clover hides it.  I can show hidden volumes and select it, then it boots perfectly fine but this is a manual process.  Seems there is not a way to configure automatic / default boot of a device it clover decides to hide it.

What is clover: clover is that with which this GitHub repo and issue tracker is associated.

Thanks.

denniskalpedis, 2022-11-08:
I have the same issue, when i boot into Clover it shows no drives, they are all hidden. I have to hit f3 and select my nvme drive and boots normal. Haven't found a way to auto select it.

SergeySlice, 2022-11-21:
If a drive is hidden then it is not entry to enter so cannot be bootable.
You have to find why the entry is hidden. It is not automatic. It is user's wish.

denniskalpedis, 2022-11-21:
How can i unhide the drive?
I can't speak for the OP, but I haven't hid any drives. I downloaded the image, set it up on USB and boot to it and no matter what I do it is set to hidden. I have never chosen to hide it, and if there is some setting to basically unhide all drives i would do it.
I have tried several things when I first set it up about a year ago.

The setup I have is USB thumbdrive with Clover and a PCI-E NVME adapter with my OS. 

and my process is boot to USB drive automatically, clover comes up and I have to hit f3 and select my drive to boot. I have tried manually editing the options file on the usb but because it is hidden it won't boot. and nothing i have done previously has changed that.

lostboy7, 2022-11-21:
Yes, clover does hide some boot volumes on its own / by default.  At the time I posted this I had identified the code that does it and could potentially patch it myself except I couldn’t figure out how to set up a build environment to rebuild clover.

In the interim time I came up with another solution (not involving clover) so I’m all set; but, I still would suggest this issue get fixed for others.

denniskalpedis, 2022-11-21:
@lostboy7 What was your fix? I would love a fix to auto boot my nvme.

SergeySlice, 2023-01-05:
Don't hide a drive you want to boot.

closed 2023-01-05 by SergeySlice

lostboy7, 2023-01-06:
Did this get fixed?  It still appears the latest version is auto hiding drives and thus not allowing boot from them.

Which commit resolves this?

denniskalpedis, 2023-01-06:
It is not fixed. From what I understand it was closed with the message of "Don't hide a drive you want to boot." But again, I am not hiding it, it is hidden by default for some reason, and nothing I do will unhide it.

So far I still need to manually hit f3, select my drive to boot.

lostboy7, 2023-01-06:
Oh, I gotcha.  I assumed that message meant it was fixed since this problem report was very clear that the issue has to do with automatic hiding, not something we configure.

I guess the question is, what to do.. can these be reopened, or should it be reported as a new issue?  I’m not familiar enough with this system…

Trinnik, 2023-01-26:
I have the exact same issue. Default clover install, have to press f3 to boot to my NVME drive despite not making a single configuration change from the default clover config.

grokspeed, 2023-04-10:
Came here exactly for this issue - in case anyone wonder why there could be hidden entries not hidden by user - here is one reason:  NVMe installed on PCI slot for older motherboard (mine is an AM3 MB), the bios cannot boot from that NVMe.

I actually came to Clover for addressing this use case.  Too bad it can't auto boot after power outage.  But for the most part it saves me from paying $15 for a SATA adapter.

Hopefully the team will add support for hidden volume as default soon.