CloverHackyColor/CloverBootloader #574: clover 5150 does not inject hw.target in macOS Ventura

jonasw8, 2022-11-10:
Hi! I have Monterey and Ventura in dualboot, but clover does not inject hd.target in macOS Ventura. Because of this I don't receive system updates. My SMBIOS is iMac 20,1 and HW.Target used is J185AP. My SIP Value is 0x00000000

MacOS Monterey

[image]

macOS Ventura
[image]

Thank you for your attention

SergeySlice, 2022-11-13:
Yes, HWTarget is not injected for Ventura.

closed 2022-11-13 by SergeySlice

alessaba, 2023-01-04:
What's the proper way to get OTA updates then? is there a proper way to have the hw.target variable set?

5T33Z0, 2023-01-27:
There's a better workaround for this but (requires macOS Big Sur 11.3 or newer):

- Add `RestrictEvents.kext` and  boot-arg `revpatch=sbvmm`. Frces VMM SB model, allowing OTA updates for unsupported models on macOS 11.3 or newer.

alessaba, 2023-01-27:
uhm... i can try this, but i tried with OpenCore and it injects the variable in Ventura too (and the OTA works) 
i had compatible hw targets for both my laptop and desktop pc when on Monterey.

5T33Z0, 2023-01-27:
It's not about the targets – it's just that [HWTarget no lonnger works in Ventura](https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2800185)…

alessaba, 2023-01-27:
Oh ok. I wonder how OpenCore does it then. i am pretty sure i checked and the hw.target variable was set to a T2 Enabled MacBook Pro and OTA works.

I will try with your workaround :-)