CloverHackyColor/CloverBootloader #579: Missing quirk [ForceAquantiaEthernet]

fabiosun, 2022-12-03:
Hi @SergeySlice 
from OpenCore 0.80 opencore devs added this quirk, it is useful to have working well AQC-107s based 10 Gbe network card
Clover seems to miss it and it is very useful today also in Monterey and Ventura OS
Is it possible to add in your boot loader?

0.8.0
Added ForceAquantiaEthernet for Aquantia AQtion AQC-107s based 10GbE network cards support, thx @Mieze and [@Shikumo](https://www.insanelymac.com/forum/profile/698028-shikumo/)
Thank you

jief666, 2024-01-04:
I've just added ForceAquantiaEthernet.
Can you try : [CloverX64-2024-01-04-11-37-47-bd46eef-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13828937/CloverX64-2024-01-04-11-37-47-bd46eef-jief.zip)? (or compile it yourself from master).

fabiosun, 2024-01-04:
@jief666 thank you but i have no more that system
Sorry

closed 2024-01-04 by jief666