CloverHackyColor/CloverBootloader #580: Bootmenu GUI: csr-active-config changes are not applied when switching configs.

5T33Z0, 2022-12-07:
## Problem description:
I have 2 configs for my Ivy Bridge Notebook: one for running macOS Catalina (config.plist) and one for Monterey (config_Monterey.plist). There are only 2  differences between them; the Monterey config uses `-no_compat_check` and a different value for csr-active-config (`0x867`) so the patched-in HD4000 drivers can be loaded.

When switching from `config.plist` to `Monterey.plist` from the bootmenu the value for csr-active config is not applied. 

Used Clover build: https://github.com/CloverHackyColor/CloverBootloader/actions/runs/3584745686

SergeySlice, 2022-12-08:
I will check.

5T33Z0, 2022-12-09:
Hmm, doesn't work for me. Maybe it's because I am using Bootloader Chooser to pick Clover? I have to test it again with Clover's bootx64.efi file.

SergeySlice, 2023-01-05:
Resolved?

closed 2023-01-05 by SergeySlice