CloverHackyColor/CloverBootloader #586: clover There is no SecureBootModel options

jackacc, 2023-01-31:
Use OpenCore patch - Legacy - Patcher
> can search download on making [OpenCore - Legacy - Patcher] (https://github.com/dortania/OpenCore-Legacy-Patcher)
3.1 shut down the SIP, otherwise is not permissions to install:
- method: run into RecoveryOS "csrutil disable" and "csrutil authenticated - root disable")
- method 2: EFI - > OC - > config. The plist - > Nvram - > 7 c436110 - AB2A - 4 bbb-minus A880 FE41995C9F82 - > for FF0F0000 CSR - active - config values
3.2 set to start or add parameter amfi_get_out_of_my_way = 1 and ipc_control_port_options = 0
- the EFI - > OC - > config. Plist - > Nvram BBB - > 7 c436110 - AB2A - 4 - A880 - FE41995C9F82 - > add amfi_get_out_of_my_way bootargs = 1
- the EFI - > OC - > config. Plist - > Nvram BBB - > 7 c436110 - AB2A - 4 - A880 - FE41995C9F82 - > add ipc_control_port_options bootargs = 0
- the EFI - > OC - > config. Plist - > Misc - > Security - > SecureBootModel: set the disable
> don't open the patch button gray, not click
[image]
[image]

5T33Z0, 2023-02-08:
1. RTVariables/HWTarget = SecureBootModel. Ignore it
2. HWTarget= does currently not work in Clover. Here's a Workaround to enable updates: https://github.com/5T33Z0/Clover-Crate/tree/main/RtVariables#hwtarget
3. Set boot-arguments instead of NVRAM parameters.
4. RTVarialisables/CsrActiveConfig: Clover uses Big Endian instead of Little Endian: `0xFEF`
5. Your actual issue issue that you don't heve Internet. Fix that first so the Patcher can do its thing!

SergeySlice, 2023-07-20:
Since Sonoma OpenCore has no more advantage over Clover. There is no Software Update for T2 models no matter of SecureBoot options of HWTarget. The OC victory came to logical End.

closed 2023-07-20 by SergeySlice

5T33Z0, 2023-07-20:
LOL