CloverHackyColor/CloverBootloader #587: Quick Question

mattdrepo, 2023-02-01:
Clover has an association with hackintosh.  But can it be used to create, disable and be more streamlined with regards to dsdt and real apple hardware and other systems (linux, *bsd, windows).  I would ideally like to modify apple's dsdt output to remove/alter  hardware for greater compatibility with sleep on linux.  Which was a thing the other way around with sleep on PC's under osx86.

SergeySlice, 2023-02-18:
In theory yes. Become to be a developer and make your own modification to the bootloader for your purpose.
First of all, if you want to modify DSDT for Linux then you should make Clover to inject modified DSDT into Linux. Now it is disabled.

SergeySlice, 2023-03-24:
No activity