CloverHackyColor/CloverBootloader #591: why  display [  Boot Microsoft EFI Boot from___________________ pdx  ]

2021smk, 2023-03-12:
in the boot menu it display  such as  Boot Microsoft EFI Boot from___________________ pdx or Boot Microsoft EFI Boot from___________________ pdp, it's hard to choose. what should do to solve this problem

SergeySlice, 2023-03-17:
You should provide preboot.log to see what is happen.

2021smk, 2023-03-18:
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/11007172/preboot.log)

SergeySlice, 2023-03-24:
It looks like you named partitions by non-ACSII characters which is not reproducible by Clovers fonts.
1:166  0:000  - [03]: Volume: PciRoot(0x0)\Pci(0x2,0x0)\Pci(0x0,0x0)\NVMe(0x1,03-10-84-00-03-8E-E3-8C)\HD(1,GPT,55F3B36D-2476-4FBC-88BB-251634C74CD6,0x800,0x32000)
1:166  0:000          Result of bootcode detection: bootable Windows (vista,win)
1:167  0:001            label : 꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯꾯桰つX

SergeySlice, 2023-03-24:
You may use vector theme and an svg font with all needed characters. See WIKI.

2021smk, 2023-03-30:
i read WIKI, i can't make a svg themes, so i  try some themes and fonts from the web.  do not work。
So  i choose the correct boot as default，it not affect use.
still thank you!