CloverHackyColor/CloverBootloader #594: Support DefaultVolume to boot hidden volume - (or re-open Issue #565)

grokspeed, 2023-04-10:
Came here exactly for this issue - in case anyone wonder why there could be hidden entries not hidden by user - here is one reason: NVMe installed on PCI slot for older motherboard (mine is an AM3 MB), the bios cannot boot from that NVMe.

(I believe #565 was closed under the assumption that a user can simply un-hide a volume?)

I actually came to Clover for addressing this use case. Too bad it can't auto boot after power outage. But for the most part it saves me from paying $15 for a SATA adapter.

Hopefully the team will add support for hidden volume as default soon.

SergeySlice, 2023-04-18:
May be you asked about booting from unsupported hardware?

bergander, 2023-06-28:
@SergeySlice  It boots perfectly fine after unhiding by pressing F3. So the hardware is supported and working, it's just hidden by default in Clover.

So the question is can Clover stop automatically hide volumes on a NVMEe in a PCI slot? Or can a hidden volume be the default volume to make it auto boot?

grokspeed, 2023-06-29:
Correct.  

The former.  

Although option to combine it with the latter is even better.

Let me add more color to the use case.  This is a media PC so everytime after power outage wife would complain the complexity to reboot (aforementioned situation - MB bios lacking NVMe boot and Clover as a workaround).  

wife cannot understand/remember the navigation on Clover.  Adding to the trouble is all my keyboards are default to non-Fn key row when not holding the shift-key.  Hope this better illustrate where I am coming from.

This little requested feature will save another PCIe gen2 MB from going into the landfill. :)

kelvintechie1, 2023-07-23:
Throwing my hat into the ring for this feature as well. Using Clover on a USB drive to boot from VMware ESXi using an NVMe SSD installed in one of the PCIe slots of a Dell PowerEdge server for my home lab. Hardware is fully operational, the only problem is that the PowerEdge UEFI does not have the proper NVMe drivers for the consumer SSDs that I am using. Since Clover does, I'm using Clover to essentially slipstream the NVMe drivers into the boot chain.

This is what preboot.log looks like:
38:466  0:000  - [07]: 'EFI'
38:467  0:000        AddLoaderEntry for Volume Name=EFI, idx=1
38:467  0:000        OSVersion= 
38:467  0:000        IconName=unknown comma=-1 size=7
38:467  0:000        Full=unknown
38:467  0:000        Loader entry created for 'PciRoot(0x0)\Pci(0x2,0x0)\Pci(0x0,0x0)\NVMe(0x1,01-0D-CD-12-67-4C-E0-00)\HD(1,GPT,F1E5AD9E-5867-4F9F-B67D-1592069C28E0,0x40,0x32000)\EFI\BOOT\BOOTX64.efi'
38:467  0:000        Menu entry added at index 1
38:467  0:000       hiding entry because DiskKind is DISK_KIND_INTERNAL: \EFI\BOOT\BOOTX64.efi

As shown below, the "Hide?" portion of my config.plist is an empty array, signifying that it isn't a config in the XML causing it to hide these drives manually.

<key>Hide?</key>
<array>
</array>

TritonB7, 2023-08-15:
Same here, booting hidden volumes would be nice.