CloverHackyColor/CloverBootloader #595: Can Clover boot Legacy on UEFI Class 3?

2W10, 2023-04-11:
Hi,

Can the Clover bootloader boot Legacy on UEFI Class 3? If so, how?

tech128, 2023-04-15:
No that is not the intended goal of clover.  Thru its uefi emulation code, it can boot EFI on legacy hardware.  Not the other way around.

The only 3 ways I know of loading csm on a class 3 system would be

1. Mod the bios to add csm support - very difficult and I haven't heard of anyone succeeding yet
2. Port seabios as an EFI app - again very difficult and I haven't heard of anyone succeeding yet
3. Using biefircate to load 16bit or 32bit apps from 64bit uefi.  Still doesn't work, but gets the closest so far

DI555, 2023-05-02:
it were a news, about the grub can do this..  was telling that it was so called "bug" in grub that can provide this kind of boot..
so, may be possible to port this grub’s "bug" into clover?   (*^‿^*)

SergeySlice, 2023-06-03:
Become a developer and do this.

closed 2023-06-20 by SergeySlice