CloverHackyColor/CloverBootloader #597: Dell Inspirion 1525: Clover 5151 "out-of-the-box" doesn't show UI

pankoz, 2023-04-27:
Hi!
Description: I'm have old laptop Dell Inspiron 1525 with Win 7 on board (installed as MBR). My disks are:
standard HDD;
additional M2 NVMe SSD inserted into UWP slot on MB (with adapter. It works under Win 7 and I can use it for files)
So, I'm searching for how-to-make-ssd-loadable. I'm trying to use USB flash drive with Clover r5151, the flash is prepared by BDU v2.1.030. My issues are:
1. There is no UI on Clover loading - it just shows me gray screen with 5 points
Continue
Select Language
Boot Manager
Device Manager
Boot Maintenance Manager
2. Older versions like r5146 for example - shows me UI but don't starts existing Win7, although I'm able to choose it
3. Are there some exceptions for Inspiron 1525 in Clover config? (I saw in docs some notes about this but can't find any details.)
4. Is there some Clover Configurator for Windows?

pankoz, 2023-04-27:
Screen photo:
[image]

pankoz, 2023-05-20:
Already non-actual questions:
2. Older versions like r5146 for example - shows me UI but don't starts existing Win7, although I'm able to choose it
I've found in Clover docs that Clover works bad with legacy bios-loaded systems and this feature is no more maintainable.
So if you're trying to use Clover with your PC/laptop then just create and use EFI Windows installers. Non-efi installers will not be loaded without magic :-) 
3. Are there some exceptions for Inspiron 1525 in Clover config? (I saw in docs some notes about this but can't find any details.)
In fact, I've adjusted the set of efi files for loader. I hadn't done something special with config.plist, just used Clover r5146.
If necessary, everybody can look into my repo "CloverSettings_Inspiron_1525" with all Clover loader files with my working config.

But the 1st and 4th are still actual. Config.plist for r5151 was modified for GUI usage according to Clover docs, I've re-checked this. A bug?

SergeySlice, 2023-06-03:
I also had Dell 1525 five years ago and kept here 10.6 booted by Clover 3xxx.
If you see screen with five entries, "Continue", "Boot Manager"... etc then your boot file doesn't found /EFI/CLOVER/CLOVERX64.EFI. Check if it exists.
2. Clover still works fine with legacy systems. I regularly checks this. Including 5151.
3. Windows should boot.
4. Inspiron 1525 have to be tuned by config.plist. And I had very patched DSDT for it.  But I lost these files.

SergeySlice, 2023-06-20:
abandoned

closed 2023-06-20 by SergeySlice