CloverHackyColor/CloverBootloader #600: clover not booting

pterotronic, 2023-05-11:
I got the computer to boot to clover. However, no matter what disk you pick in the clover menu, it does not boot. If i boot directly to c:, it boots fine. If I boot to c: from the cover menu. If fails. You just get a blank screen with a cursor blinking in the upper left corner.

I want to boot to nvme in pci-e 3 slot. But if it won't even boot to c:, I think there's something wrong that needs to be fixed first.

thanks for any help w this.

SergeySlice, 2023-06-04:
You should have NVMExpressDxe.efi and target disk must have what to boot.

SergeySlice, 2023-06-20:
abandoned

pterotronic, 2024-01-30:
I have NVMExpressDxe.efi in usb root, and in usb EFI/boot and usb efi/clover subdir. I also have it on the nvme target disc in root, root/windows, root/system32 and/root system. The nvme is a clone of the c: system disk. Again, the nvme is perfectly functional - shows up in disk manager, file explorer and even the clover boot menu. It's just that when I select anything in clover menu, whether c: drive or nvme, I just get black screen.

pterotronic, 2024-01-30:
Correction: The first few times i tried it, I could not even boot to c: from clover menu. But I did reinstall clover to USB, it might have been a newer version now. I don't remember. In any case, I can boot to C: from clover menu now. But still can't boot to NVME from clover - just get black screen.