CloverHackyColor/CloverBootloader #603: Clover disobeys Efi Version set in config.plist/SMBIOS

LeeBinder, 2023-05-26:
Clover: latest 5151 + CLOVERX64.efi 5151a

Screenshot from Clover Configurator (sensitive data blurred out):

[image]

Screenshot from [SilentKnight](https://eclecticlight.co/lockrattler-systhist/):

[image]

Same "ghostly" Efi Firmware Version also displayed in "About this Mac > System Information", so it's not a SilentKnight issue.

When I boot OpenCore with the "same" (or rather translated/ converted) PlatformInfo settings in OC config.plist, SilentKnight displays the updated Efi Firmware Version.

Below the SMBIOS section as plain text (sensitive data *****ed out):

<details>
  <summary>click to expand</summary> 

  ```javascript
<key>SMBIOS</key>
	<dict>
		<key>BiosReleaseDate</key>
		<string>03/04/2023</string>
		<key>BiosVendor</key>
		<string>Apple Inc.</string>
		<key>BiosVersion</key>
		<string>MBP154.88Z.F000.B00.2303040348</string>
		<key>Board-ID</key>
		<string>Mac-53FDB3D8DB8CA971</string>
		<key>BoardManufacturer</key>
		<string>Apple Inc.</string>
		<key>BoardSerialNumber</key>
		<string>*******</string>
		<key>BoardType</key>
		<integer>10</integer>
		<key>BoardVersion</key>
		<string>MacBookPro15,4</string>
		<key>ChassisAssetTag</key>
		<string>Asset Tag#</string>
		<key>ChassisManufacturer</key>
		<string>Apple Inc.</string>
		<key>ChassisType</key>
		<string>0x9</string>
		<key>EfiVersion</key>
		<string>1968.120.12.0.0</string>
		<key>ExtendedFirmwareFeatures</key>
		<string>0x00000008FD8FF426</string>
		<key>ExtendedFirmwareFeaturesMask</key>
		<string>0x00000008FF9FFF3F</string>
		<key>Family</key>
		<string>MacBook Pro</string>
		<key>FirmwareFeatures</key>
		<string>0xFD8FF426</string>
		<key>FirmwareFeaturesMask</key>
		<string>0xFF9FFF3F</string>
		<key>LocationInChassis</key>
		<string>Part Component</string>
		<key>Manufacturer</key>
		<string>Apple Inc.</string>
		<key>Mobile</key>
		<true/>
		<key>PlatformFeature</key>
		<string>0x32</string>
		<key>ProductName</key>
		<string>MacBookPro15,4</string>
		<key>SerialNumber</key>
		<string>**************</string>
		<key>SmUUID</key>
		<string>02A93CB0-7C*****************-C7058E7EB80E</string>
		<key>Version</key>
		<string>1.0</string>
	</dict>
  ```
</details>

Is there some other key whose string needs to be updated?

LeeBinder, 2023-05-29:
Insights have been shed [here](https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2805254) ff. on insanelymac.

SergeySlice, 2023-05-31:
Resolved with 5152

closed 2023-05-31 by SergeySlice