CloverHackyColor/CloverBootloader #606: AMD AM5 platform support in Clover Bootloader

fabiosun, 2023-06-05:
Hi @SergeySlice 
Is it possible to apply this patch also in Clover Bootloader?
https://github.com/acidanthera/OpenCorePkg/pull/396/files

with this patch is possible to use AM5 (AMD platform) without audio distortion

Clover boot loaders boots fine with AM5 platform but there are audio problem (as before of above patches happened also in Opencore boot loader)

Attached a couple of Clover logs (R5149 and R5152) both booting fine (with audio problem)

thank you
[2023-06-04_10-08_CLOVERX64.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/11649371/2023-06-04_10-08_CLOVERX64.efi.log)
[2023-06-04_14-36_CLOVERX64.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/11649372/2023-06-04_14-36_CLOVERX64.efi.log)

SergeySlice, 2023-06-20:
I am very sorry but the patch is already in current Clover project, OpenCore subproject.

closed 2023-06-20 by SergeySlice

fabiosun, 2023-06-20:
@SergeySlice 
It doesn’t work
AM5 users can’t use Clover bootloader then!