CloverHackyColor/CloverBootloader #608: esxi bios mode iso boot issue R5152

kcj83, 2023-06-09:
Can someone please point me in the right direction with diagnosing/debugging this BIOS mode iso boot issue shown below on esxi8.0c virtual machine? It just hangs directly after bios initialization and booting from the latest clover iso.
[image]
[image]

kcj83, 2023-06-09:
vmware.log attached
[vmware.log](https://github.com/CloverHackyColor/CloverBootloader/files/11709500/vmware.log)

SergeySlice, 2023-06-20:
Sorry, I am understanding zero in VMWare. Become developer and tell us what to do with vmware.

closed 2023-06-20 by SergeySlice