CloverHackyColor/CloverBootloader #611: Difficulties booting Clover on a Legacy BIOS system

Brayan2005, 2023-06-20:
I downloaded clover r5152 and put it on a USB drive using rufus. (I simply used the default settings, MBR partition scheme for UEFI target system)
When trying to boot it, I simply got a black screen with a blinking cursor. I read that this could be because of conflicting boot devices, but that's not possible, since it's the only bootable device and removing it gives me an error message saying no bootable devices have been found.
I also tried adding other bootable devices, to see if clover was dependent on it, since it's a bootloader, but that didn't change anything.
I then tried some troubleshooting steps until I got the idea to change my rufus parameters.
I switched the partition scheme from MRB to GPT. After this change I started seeing this screen :
[image]

I checked my firmware settings for anything mentionning "Legacy mode", but I only found "USB Legacy mode" and turning that off makes me unable to boot from USB at all. (so I turned it back on of course)

I then checked rufus again, to follow the instructions :
-change partition scheme back to MBR
-and change the target system from UEFI to BIOS
Unfortunately, the only option I get for the target system is UEFI.
BIOS simply does not show.
No matter if I select MBR or GPT, it only lets me select UEFI.

I have read on the GitHub page that clover should work on both UEFI and BIOS systems so I guess I'm simply doing something wrong. So I started asking around on forums and discord servers, but no one was able to help me.
So I finally came to ask for help here.

What do I need to do, to boot clover on this device?
It's an ASUS P7P55D-E by the way.

daveymg-nz, 2023-06-25:
I ran into this same problem with not being able to boot the usb after writing with Rufus in MBR partition scheme. This is on an Asus A55M-A mobo.

After rewriting the USB stick in GPT scheme I had to change the PC Bios settings to disable CSM mode and changed the UEFI boot setting from Windows OS to Other OS. The Clover USB then booted successfully.

After double clicking on the device icon on the Clover boot screen I then got a blank screen and the PC Bios has failed to start again since. I don't know why that's happened yet.

SergeySlice, 2023-06-28:
> > I don't know how rufus works and I don't want to know this. I have to recommend you to read [WIKI](https://github.com/CloverHackyColor/CloverBootloader/wiki) about installation.
> 
> I read the "Install for legacy boot" part, but it only seems to be explaining how legacy boot works, which I don't understand much of, then it tells me how to format the drive which is nice, but not how to install clover onto it. Then I see command lines, as if I'm reading some Linux turorial, but I obviously can't type any commands into my firmware, right? As far as I know we only use command lines in an Operating System. I also saw multiple mentions of MacOS, I surely don't need MacOS to install Clover("for macOS, Windows and Linux"), right? Anyways it didn't tell me how to install clover, maybe I'm just stupid, I don't know.

Use Clover.pkg to do automatic installation. You may choose some old version of the package and then replace one file CloverX64.efi to the newest one.

homeinterface, 2023-07-09:
I was trying to have this boot manager working on BIOS-only PC laptop of 2009 / Dell/Windows/Debian.

 It is not impossible but it seems also something buggy code hapend since a 2019 release - the release mentioned somewhere here as working (together with relevant BDU tool) .

It is clear that  only a single disk on my system is allowed to boot Clover. While having another disk together within the system makes Clover failed to start.

I have managed to install Clover manualy onto a usb pendrive using Debian Live booted from  the second pendrive. I had to use "7" version of the boot code applied to the first pendrive according to the install description. Only diffrencerence is that Linux has  "/dev/sda" naming convention for the disks attached.

To boot Clover only one disk has to be attached to the BIOS PC. Yes you cannot do much but at least Clover shows up.

To be continued...

SergeySlice, 2023-09-04:
Check new clover since commit 90b68db32

Brayan2005, 2023-09-04:
> Use Clover.pkg to do automatic installation. You may choose some old version of the package and then replace one file CloverX64.efi to the newest one.

I read that pkg files are a MacOS thing and I have never used MacOs before, I don't even know how I would get it, except by buying a mac maybe, which I will definitely not do.
I mainly use Windows 10.
I thought I could install Clover onto a PC, that has no OS yet. So I can use Clover with NVMe drivers to install an OS (Windows 10) onto an NVMe SSD (because the motherboard doesn't have NVMe support).
Many tutorials told me to do this, but like I said, I'm having issues.

SergeySlice, 2023-12-02:
Looks like abandoned. I see no Clover issue. I see user can't install Clover.

closed 2023-12-02 by SergeySlice