CloverHackyColor/CloverBootloader #612: Issue when building Clover

BokBoy, 2023-06-20:
Hi there, I have an issue when I am building Clover. It seems to be a redefine of an variable but I may as well leave it to you guys. 
Here is the output;
"In file included from CommonLib.c:18:
In file included from ../Common/CommonLib.h:12:
In file included from ../Include/Common/UefiBaseTypes.h:13:
In file included from ../Include/Common/BaseTypes.h:17:
../Include/AArch64/ProcessorBind.h:58:11: error: 'UINT8_MAX' macro redefined [-Werror,-Wmacro-redefined]
  #define UINT8_MAX 0xff
          ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/stdint.h:107:9: note: previous definition is here
#define UINT8_MAX         255
        ^
1 error generated.
make[2]: *** [CommonLib.o] Error 1
make[1]: *** [Common] Error 2
make: *** [Source/C] Error 2"
If you can, can you please fix or tell me how to do this?? Thanks!!

SergeySlice, 2023-06-20:
AArch64??? Are you really want to compile Clover to aarch64?
It seems developer tools on MacBook Air M2 may compile only for aarch64.

BokBoy, 2023-06-20:
Oh yeah, I also had an error about AArch64. It also said earlier than the error, it was getting the AMD64 tools. So things like the GCC, etc, it said finding for AMD64. 
Finally, It was getting more outputs than URLs, is that normal or not good?

SergeySlice, 2023-06-28:
I think we may do the follow
~~~~
#ifndef UINT8_MAX
  #define UINT8_MAX 0xff
#endif
~~~~

to avoid the redefinition.
But Clover is not ready to work on aarch64 machines.

BokBoy, 2023-07-03:
Hi there, Thanks for the answer.
It turns out that Clover Bootloader will not anyways support the newer Apple ARM Chipset due to the change from EFI or UEFI to the new ISC boot. This means that apple decided to ditch the choice of allowing a EFI Boot and required you to use the new Apple APFS ISC. This means that Clover will not work on it.
This means the end of my manual attempt of dual booting windows on the M2 Mac so this will no longer work. 
Thank you for your assistance and my tip is - if somebody asks for Apple silicon usage, do not let them as it will not work.
Thank you @SergeySlice for the assistance.
Thanks, BokBoy

closed 2023-07-03 by BokBoy