CloverHackyColor/CloverBootloader #617: Problem with loading Clover

PlushPersik, 2023-07-25:
I wanted to put a hackintosh on my PC. But for some reason, when I did everything right, namely: I inserted a USB flash drive, downloaded BDUtility, wrote the latest version Clover to the USB flash drive, then write the .rdr image with macOS High Sierra to the free partition through the R-Drive image program.

I rebooted the PC, booted from the flash drive, then a white flashing bar appeared in the up right area. Then the number 6 appeared, and a screen with red letters came out.

My pc:
Intel Xeon x3460
GeForce GTX 750 Ti
8 GB RAM 1333 MHz
No any boost
Hard disk 320 GB
Legacy bios

PlushPersik, 2023-07-25:
[image]

SergeySlice, 2023-09-05:
Choose different boot6 or boot7 file and rename it to boot.
Replace the file on your USB drive.

SergeySlice, 2023-10-14:
abandoned

closed 2023-10-14 by SergeySlice