CloverHackyColor/CloverBootloader #620: NVMe over USB Type-c Fails to boot.

xyz32, 2023-07-29:
I got an USBc hub with an NVMe port.

I installed an OS on the nvme, and if I use the but with my 1 year old work dell laptop over the usb-c port everything works.

When I attache this hub to my old desktop (running an Asus 990FX r2) I can only boot the nvme rive if I attach it to an USB-A 2.0 or lower.
I just got an USB-A 3.0 to USB-C dongle and neither the original BIOS nor Clover can boot off of it. I also got an 20Gbps PCIe to USB-C card and that doesn't boot either.

The setup looks sort of something like this:
[https://pasteboard.co/N5MSm0zpc3Dm.jpg](https://pasteboard.co/N5MSm0zpc3Dm.jpg)

Will Clover be able to boot this setup (either over the PCIe card or the USB-A 3.0?

This is the preboot log for the 3.0 dongle:
https://pastebin.com/HLVsjggC

And the logs for the working config over USB-A 2.0:
https://pastebin.com/dj18cZt5

For Clover drivers I am only using
NvmExpressDxe.efi
EnglishDxe.efi

I tried with enabling all the dex files but still nothing.

The OS I am booting is Fedora 38.

Thanks.

SergeySlice, 2023-10-23:
NvmExpressDxe assumes PCIe bus but not USB.

closed 2023-10-23 by SergeySlice

eugenesvk, 2024-06-25:
Am I correct that this also means that there is no support for bootable USB3 drives connected  via PCEe extension cards? Thanks!

xyz32, 2024-06-25:
> Am I correct that this also means that there is no support for bootable USB3 drives connected via PCEe extension cards? Thanks!

Most probably, from the hardware point of view the PCIe usb will just show up as another USB hub.

If you are trying to boot into linux, you can use another linux install as a boot-loader. I wrote this some time ago: [](https://github.com/CloverHackyColor/CloverBootloader/discussions/621#discussioncomment-6632825)

eugenesvk, 2024-06-26:
Unfortunately I was trying to boot Windows :(