CloverHackyColor/CloverBootloader #625: Block IOSkywalkFamily under Sonoma

anonymous-writer, 2023-08-12:
Hello everyone and thanks for working on the Clover Bootloader.

Is there a way to block the kext IOSkywalkFamily under Sonoma so that the Ventura version of the IOSkywalkFamily can be loaded?
The info plist patch doesn't want to work and the kext always loads.

[image]

[image]

SergeySlice, 2023-09-03:
Made since commit 2711170df4f60b2ae5aa20add3e00f35cf57b7e5

closed 2023-09-03 by SergeySlice

anonymous-writer, 2023-09-10:
I have seen and tested this. Unfortunately it doesn't work and gets stuck at the picture as soon as you set the BlockSkywalk to true. 

[image]