CloverHackyColor/CloverBootloader #626: Clover can't open TRIM for SATA SSD in Venrtura and Sonoma

gujiangjiang, 2023-08-29:
I'm using Clover and OpenCore and I find the clover can't open trim for my sata ssd but the OpenCore with no problem.

[image]
Here is the Sonoma in OpenCore.

[image]
Here is the Sonoma in Clover.

[image]
With Clover config I have checked the [ThirdPartyDrive] option same with OpenCore config but have no use in Clover but it works in OpenCore.

I hope to fix this bug to make  [ThirdPartyDrive]  option works in Clover too.

Thanks.

gujiangjiang, 2023-08-30:
I find OpenCore have solve this problem on Mar 30.

https://github.com/acidanthera/OpenCorePkg/pull/438/commits/e9f0071575f8914f1cf47f2adb3a039e247f92e1

Please update the source code from the latest OpenCore.

Thanks.

SergeySlice, 2023-09-03:
Done! Check please if it works or more updates needed.

gujiangjiang, 2023-09-04:
@SergeySlice 
Thanks for your work.
I tried with new beta version for Clover in GithubAction and the trim for Ventura and Sonoma works for me with [ThirdPartyDrives].

Thanks again.

gujiangjiang, 2023-09-04:
@SergeySlice 
hi slice.
I test with Ventura and Sonoma but I find a bug in WIFI.
The [com.apple.iokit.IOSkywalkFamily] was disabled by default so when I boot into Sonoma ,my intel wifi can't load anymore.
PS:I have no [IOSkywalkFamily] in config, my config was old with the latest released clover, So the new option [BlockSkywalk] works default in Sonoma? I don't think this is a good read.

SergeySlice, 2023-09-05:
Now BlockSkywalk is set to false by default.