CloverHackyColor/CloverBootloader #634: IOSkywalk blocking >> kernel panic, Clover 5155 latest commit 0ea0020

perez987, 2023-09-26:
Good night, when trying to set Clover to get back Broadcom wifi on Sonoma, every time IOSkywalk is blocked I get a KP.

In the DEBUG file, I see that the 3 required kexts are properly injected:

```bash
127:901  0:018  ->Extra kext: Other\IOSkywalkFamily.kext (v.1.0)
127:919  0:018  ->Extra kext: Other\Lilu.kext (v.1.6.6)
127:937  0:017  ->Extra kext: Other\AMFIPass.kext (v.1.3.1)
127:956  0:018  ->Extra kext: Other\AppleALC.kext (v.1.8.3)
127:974  0:018  ->Extra kext: Other\IntelMausi.kext (v.1.0.8)
127:992  0:018  ->Extra kext: Other\RestrictEvents.kext (v.1.1.3)
128:010  0:017  ->Extra kext: Other\WhateverGreen.kext (v.1.6.5)
128:058  0:018  ->Extra kext: Other\IO80211FamilyLegacy.kext (v.1200.12.2b1)
128:077  0:018      |-- PlugIn kext: Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext (v.1400.1.1)
128:095  0:018  ->Extra kext: Other\FakeSMC.kext (v.3.5.5)

128:283  0:018  Bridge kext to OC : Path=Lilu.kext
128:304  0:020  info plist path: Kexts\Other\Lilu.kext\Contents\Info.plist
128:325  0:021  InfoPlistPath=Kexts\Other\Lilu.kext\Contents\Info.plist
128:347  0:021  assign executable as 'Contents\MacOS\Lilu'
128:366  0:019  Bridge kext to OC : Path=FakeSMC.kext
128:387  0:020  info plist path: Kexts\Other\FakeSMC.kext\Contents\Info.plist
128:422  0:035  InfoPlistPath=Kexts\Other\FakeSMC.kext\Contents\Info.plist
128:445  0:023  assign executable as 'Contents\MacOS\FakeSMC'
128:466  0:020  Bridge kext to OC : Path=IOSkywalkFamily.kext
128:487  0:020  info plist path: Kexts\Other\IOSkywalkFamily.kext\Contents\Info.plist
128:510  0:023  InfoPlistPath=Kexts\Other\IOSkywalkFamily.kext\Contents\Info.plist
128:532  0:021  assign executable as 'Contents\MacOS\IOSkywalkFamily'
128:563  0:031  Bridge kext to OC : Path=WhateverGreen.kext
128:584  0:020  info plist path: Kexts\Other\WhateverGreen.kext\Contents\Info.plist
128:606  0:021  InfoPlistPath=Kexts\Other\WhateverGreen.kext\Contents\Info.plist
128:629  0:023  assign executable as 'Contents\MacOS\WhateverGreen'
128:740  0:110  Bridge kext to OC : Path=AppleALC.kext
128:943  0:203  info plist path: Kexts\Other\AppleALC.kext\Contents\Info.plist
129:138  0:194  InfoPlistPath=Kexts\Other\AppleALC.kext\Contents\Info.plist
129:545  0:407  assign executable as 'Contents\MacOS\AppleALC'
129:770  0:225  Bridge kext to OC : Path=AMFIPass.kext
129:998  0:227  info plist path: Kexts\Other\AMFIPass.kext\Contents\Info.plist
130:216  0:217  InfoPlistPath=Kexts\Other\AMFIPass.kext\Contents\Info.plist
130:439  0:223  assign executable as 'Contents\MacOS\AMFIPass'
130:463  0:024  Bridge kext to OC : Path=IntelMausi.kext
130:490  0:026  info plist path: Kexts\Other\IntelMausi.kext\Contents\Info.plist
130:517  0:026  InfoPlistPath=Kexts\Other\IntelMausi.kext\Contents\Info.plist
130:555  0:038  assign executable as 'Contents\MacOS\IntelMausi'
130:579  0:024  Bridge kext to OC : Path=RestrictEvents.kext
130:606  0:026  info plist path: Kexts\Other\RestrictEvents.kext\Contents\Info.plist
130:633  0:026  InfoPlistPath=Kexts\Other\RestrictEvents.kext\Contents\Info.plist
130:660  0:026  assign executable as 'Contents\MacOS\RestrictEvents'
130:816  0:039  Bridge kext to OC : Path=IO80211FamilyLegacy.kext
130:844  0:027  info plist path: Kexts\Other\IO80211FamilyLegacy.kext\Contents\Info.plist
130:871  0:027  InfoPlistPath=Kexts\Other\IO80211FamilyLegacy.kext\Contents\Info.plist
130:898  0:026  assign executable as 'Contents\MacOS\IO80211FamilyLegacy'
130:934  0:036  Bridge kext to OC : Path=AirPortBrcmNIC.kext
130:963  0:028  info plist path: Kexts\Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext\Contents\Info.plist
130:990  0:027  InfoPlistPath=Kexts\Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext\Contents\Info.plist
131:017  0:027  assign executable as 'Contents\MacOS\AirPortBrcmNIC'
```

And IOSkyWalk is blocked but after the addition of the 3 kexts:

```bash
151:205  0:045  OCAK: Patching invalid size 2EFF8 with 1456000 for com.apple.iokit.IOSkywalkFamily
151:230  0:024  OCAK: com.apple.iokit.IOSkywalkFamily got KXLD 0 0
151:254  0:024  OCAK: Excluding com.apple.iokit.IOSkywalkFamily - VirtualBase FFFFFF80027FE000, MaxSize 40000
151:291  0:036  OCAK: Found kext com.apple.iokit.IOSkywalkFamily (2C377398) from link 2C3773A0 to drop
151:315  0:024  OCAK: Erasing com.apple.iokit.IOSkywalkFamily from prelinked kext under dict index 3, plist 2CB8CE98, plist index 167
151:339  0:024  OC: Prelinked blocker (Exclude) result 0 for com.apple.iokit.IOSkywalkFamily () - Success
``

Maybe the block must happen before the addition?
My skills are low so maybe I'm asking something meaningless. But users who are testing this with Clover have KP by blocking IOSkywalk.
Thank you!!!

etyel, 2023-10-02:
Have you followed these instructions? From https://github.com/dortania/OpenCore-Legacy-Patcher/pull/1077#issuecomment-1646934494

While the project (OpenCore legacy patcher) is designed for legacy Mac hardware, we know the community is quite interested in our development of Broadcom patches. For those who wish to use the Broadcom patches on non-standard machines such as Hackintoshes, see below:

Enabling Patching support for modern Broadcom Cards
To use our current patches, you'll need to ensure the following:

System Integrity Protection is set to 0x803
csr-active-config | data | 03080000
Reset NVRAM or add csr-active-config to Delete to ensure the new variable is set

AMFI is disabled
boot-args | string | amfi=0x80
Secure Boot Model is set to Disabled

Following kexts are blocked:
com.apple.iokit.IOSkywalkFamily 
Set MinKernel to 23.0.0 to ensure patches only apply on Sonoma

Following kexts are injected:
IOSkywalk.kext
IO80211FamilyLegacy.kext

Kext has a child, AirPortBrcmNIC.kext, ensure this is injected as well
Set MinKernel to 23.0.0 to ensure patches only apply on Sonoma

Once these are injected, you can run OpenCore-Patcher's Post-Install option and root patch. On reboot, Wireless support should be restored assuming your machine was configured correctly to the above.

perez987, 2023-10-02:
@etyel 
Of course I know OCLP sonoma guidelines, successful applied in my 2 hacks.
But they are for OpenCore. And this is Clover. 
Out of curiosity, I tried to do the same thing that works in OpenCore (my usual bootloader) but with Clover. And I have encountered (like other users) the problem that I am not able to achieve the same result. This is the reason for publishing this issue.

dateno1, 2023-10-06:
Same Problem

Install Sonoma with lastest Clover All work well exclude WLan

I can't boot with 'KernelAndKextPatches - BlockSkywalk'

No root patch (i Can't boot yet)

With&Without
IO80211FamilyLegacy.kext
IOSkywalkFamily.kext

AMFIPass.kext
or
amfi=0x80

with&without
ECM-Override.kext

All same (can't boot)
---------------------
[GA-Z390-M-EFI-Ventura-231006.zip](https://github.com/CloverHackyColor/CloverBootloader/files/12832266/GA-Z390-M-EFI-Ventura-231006.zip)
EFI Attatched

---------------------
Harware
CPU : i7-9700
RAM : 32G*2
Board : GA-Z390-M
VGA : RX5700XT
WLan : BCM94360CS2
with Titan Ridge

anonymous-writer, 2023-10-06:
I've also tried everything possible. Always the same problem right at startup.
https://github.com/CloverHackyColor/CloverBootloader/issues/625#issuecomment-1712841154

jonasw8, 2023-10-08:
> Same Problem
>
> Install Sonoma with lastest Clover All work well exclude WLan
>
> I can't boot with 'KernelAndKextPatches - BlockSkywalk'
>
> No root patch (i Can't boot yet)
>
> With&Without [...]

Same problem

SergeySlice, 2023-10-15:
> Good night, when trying to set Clover to get back Broadcom wifi on Sonoma, every time IOSkywalk is blocked I get a KP.
> 
> In the DEBUG file, I see that the 3 required kexts are properly injected:
> 
> ```shell
> ->Extra kext: Other\IOSkywalkFamily.kext (v.1.0)
> ->Extra kext: Other\Lilu.kext (v.1.6.6)
> ->Extra kext: Other\AppleALC.kext (v.1.8.3)
> ->Extra kext: Other\IntelMausi.kext (v.1.0.8)
> ->Extra kext: Other\WhateverGreen.kext (v.1.6.5)
> ->Extra kext: Other\IO80211FamilyLegacy.kext (v.1200.12.2b1)
>     |-- PlugIn kext: Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext (v.1400.1.1)
> ->Extra kext: Other\FakeSMC.kext (v.3.5.5)
> 
> Bridge kext to OC : Path=Lilu.kext
> info plist path: Kexts\Other\Lilu.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\Lilu.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\Lilu'
> Bridge kext to OC : Path=FakeSMC.kext
> info plist path: Kexts\Other\FakeSMC.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\FakeSMC.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\FakeSMC'
> Bridge kext to OC : Path=IOSkywalkFamily.kext
> info plist path: Kexts\Other\IOSkywalkFamily.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\IOSkywalkFamily.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\IOSkywalkFamily'
> Bridge kext to OC : Path=WhateverGreen.kext
> info plist path: Kexts\Other\WhateverGreen.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\WhateverGreen.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\WhateverGreen'
> Bridge kext to OC : Path=AppleALC.kext
> info plist path: Kexts\Other\AppleALC.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\AppleALC.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\AppleALC'
> Bridge kext to OC : Path=IntelMausi.kext
> info plist path: Kexts\Other\IntelMausi.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\IntelMausi.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\IntelMausi'
> Bridge kext to OC : Path=IO80211FamilyLegacy.kext
> info plist path: Kexts\Other\IO80211FamilyLegacy.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\IO80211FamilyLegacy.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\IO80211FamilyLegacy'
> Bridge kext to OC : Path=AirPortBrcmNIC.kext
> info plist path: Kexts\Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext\Contents\Info.plist
> InfoPlistPath=Kexts\Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext\Contents\Info.plist
> assign executable as 'Contents\MacOS\AirPortBrcmNIC'
> ```
> 
> And IOSkyWalk is blocked but AFTER the addition of the 3 kexts:
> 
> ```shell
> OCAK: Patching invalid size 2EFF8 with 1456000 for com.apple.iokit.IOSkywalkFamily
> OCAK: com.apple.iokit.IOSkywalkFamily got KXLD 0 0
> OCAK: Excluding com.apple.iokit.IOSkywalkFamily - VirtualBase FFFFFF80027FE000, MaxSize 40000
> OCAK: Found kext com.apple.iokit.IOSkywalkFamily (2C377398) from link 2C3773A0 to drop
> OCAK: Erasing com.apple.iokit.IOSkywalkFamily from prelinked kext under dict index 3, plist 2CB8CE98, plist index 167
> OC: Prelinked blocker (Exclude) result 0 for com.apple.iokit.IOSkywalkFamily () - Success
> ```
> 
> Maybe the block must happen before the addition? My skills are low so maybe I'm asking something meaningless. But users who are testing this with Clover have KP when blocking IOSkywalk. Thank you!!!

Yes, blocking must happen before the addition. I just wonder why it is not the case.

Striker770, 2023-10-18:
> Yes, blocking must happen before the addition. I just wonder why it is not the case.

So, how can it be fixed?

perez987, 2023-10-18:
Yes @SergeySlice can you fix this?
I’ve tried acting over rEFIt_UEFI/refit/main.cpp to put the block before the addition but, as you know, I’m not a coder and got errors when compiling. 
Thanks.

SergeySlice, 2023-10-18:
I looked into sources and see
`  OcKernelBlockKexts (Config, DarwinVersion, Is32Bit, CacheTypeCacheless, Context);

  OcKernelInjectKexts (Config, CacheTypeCacheless, Context, DarwinVersion, Is32Bit, 0, 0);

  OcKernelApplyPatches (Config, mOcCpuInfo, DarwinVersion, Is32Bit, CacheTypeCacheless, Context, NULL, 0);
`
So it does block first and inject second as it should be.

perez987, 2023-10-18:
Indeed, this is the expected order in OpenCore, as in Configuration.pdf:

ˋˋ
Kernel and kext changes apply with the following effective order:

- Block is processed.
- Add and Force are processed.
- Emulate and Quirks are processed.
- Patch is processed.
ˋˋˋ

This need further investigation. 
Thanks and good night.

Striker770, 2023-10-18:
@SergeySlice Is it possible to add a functionality like OC's Blocking Method in the Clover? It would be really helpful and that way it can be bit more future proof and would not need blocking option for each kext like now. I know its not easy but i can think it would be really worth it. 

Thanks

SergeySlice, 2023-10-21:
Is there Quirks->KernelCache->Auto?
What if you change it to
Quirks->KernelCache->Prelinked?

perez987, 2023-10-21:
@SergeySlice 
Quirks->KernelCache exists, it was Auto, changed to Prelinked but same result, no success.
Blocking and adding kexts (in the right order) are working but blocking happens before adding.

```swift
21:694  0:040  OCAK: Patching invalid size 2EFF8 with 1456000 for com.apple.iokit.IOSkywalkFamily
21:701  0:006  OCAK: com.apple.iokit.IOSkywalkFamily got KXLD 0 0
21:708  0:007  OCAK: Excluding com.apple.iokit.IOSkywalkFamily - VirtualBase FFFFFF80027FE000, MaxSize 40000
21:716  0:008  OCAK: Found kext com.apple.iokit.IOSkywalkFamily (2ED2D218) from link 2ED2D220 to drop
21:726  0:009  OCAK: Erasing com.apple.iokit.IOSkywalkFamily from prelinked kext under dict index 3, plist 2F543498, plist index 167
21:733  0:007  OC: Prelinked blocker (Exclude) result 0 for com.apple.iokit.IOSkywalkFamily () - Success

22:125  0:007  OCAK: com.apple.iokit.IOSkywalkFamily got KXLD 0 0
...
22:252  0:013  OC: Prelinked injection Kexts\Other\IOSkywalkFamily.kext () - Success

22:389  0:006  OCAK: Requesting __LINKEDIT for com.apple.iokit.IO80211FamilyLegacy - 277C6520 at 3EE1000
...
22:754  0:013  OC: Prelinked injection Kexts\Other\IO80211FamilyLegacy.kext () - Success

22:762  0:008  OCAK: com.apple.driver.AirPort.BrcmNIC got KXLD 0 0
...
22:861  0:010  OC: Prelinked injection Kexts\Other\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext () - Success
```

perez987, 2023-10-22:
@SergeySlice 
`Quirks->KernelCache=Cacheless` also early KP but this time IOSkywalkFamily is not blocked.
`Quirks->KernelCache=Auto` or `Prelinked` -> early KP but the kexts are well blocked and added.
This is the few lines I can see on screen at boot:

[image]

b1t0y, 2023-10-22:
> @SergeySlice `Quirks->KernelCache=Cacheless` also early KP but this time IOSkywalkFamily is not blocked. `Quirks->KernelCache=Auto` or `Prelinked` -> early KP but the kexts are well blocked and added. This is the few lines I can see on screen at boot:
> 
> [image]

Hi, 
I just wondrering if you need to put the min kernel to 23.0.0 to work on ioskywalkfamily and other kext. I dont know if min kernel exist in clover though.  Coz in OC i get KP when i dont set the min kernel to 23.0.0.  

Do you mind sharing how did you put the patches in clover? I want to try it on one of my hackintosh with clover and broadcom 94352hmb. Thanks.

perez987, 2023-10-22:
@b1t0y 
I don't know how to set min kernel to a kext on Clover. But we can put kexts into the macOS related folder, e.g. `EFI/Clover/Kexts/14` to be loaded only on Sonoma.

Changes I make to Clover are:

- copy IOSkywalkFamily.kext and IO80211FamilyLegacy.kext into EFI/Clover/Kexts/14
- `KernelAndKextPatches` >> `BlockSkywalk=True`

But so far I don't know of any user having success recovering Broadcom wifi with Clover.

SergeySlice, 2023-10-23:
Yes, BlockSkywalk assumes minkernel=23.

perez987, 2023-10-23:
Good morning. I give up for now.
In the debug log It is evident that blocking, loading and ordering kexts is ok. But I still have early KP and I don't know why.
Anyway, it has helped me reconnect with Clover and recover partially forgotten knowledge.

Note: code with cuts.

```
0:106  0:001  === [ Starting Clover ] =========================
0:106  0:000  Starting Clover revision: 5155 (master, commit 0ea0020c4) on American Megatrends EFI
0:106  0:000  Build id: 20230906192706-0ea0020-5155-dirty

# kexts added from the right folders
16:694  0:008  AddKextsInArray from Other
16:702  0:007  ->Extra kext: Other\Lilu.kext (v.1.6.7)
16:713  0:010  ->Extra kext: Other\AppleALC.kext (v.1.8.4)
16:721  0:008  ->Extra kext: Other\IntelMausi.kext (v.1.0.8)
16:729  0:008  ->Extra kext: Other\USBToolBox.kext (v.1.1.1)
16:745  0:008  ->Extra kext: Other\WhateverGreen.kext (v.1.6.6)
16:753  0:008  ->Extra kext: Other\IntelCPUMonitor.kext (v.1.2.6)
16:761  0:008  ->Extra kext: Other\FakeSMC.kext (v.3.5.5)
16:769  0:007  ->Extra kext: Other\UTBMap.kext (v.1.1)
16:780  0:010  ->Extra kext: Other\RadeonMonitor.kext (v.1.3.5)
16:790  0:009  AddKextsInArray from Off
16:798  0:008  AddKextsInArray from 14
16:806  0:008  ->Extra kext: 14\IOSkywalkFamily.kext (v.1.0)
16:814  0:007  ->Extra kext: 14\IO80211FamilyLegacy.kext (v.1200.12.2b1)
16:822  0:007      |-- PlugIn kext: 14\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext (v.1400.1.1)

# IOSkywalkFamily blocked
24:239  0:042  OCAK: Patching invalid size 2EFF8 with 1456000 for com.apple.iokit.IOSkywalkFamily
24:248  0:008  OCAK: com.apple.iokit.IOSkywalkFamily got KXLD 0 0
24:258  0:010  OCAK: Excluding com.apple.iokit.IOSkywalkFamily - VirtualBase FFFFFF80027FE000, MaxSize 40000
24:267  0:008  OCAK: Found kext com.apple.iokit.IOSkywalkFamily (2ED44498) from link 2ED444A0 to drop
24:275  0:008  OCAK: Erasing com.apple.iokit.IOSkywalkFamily from prelinked kext under dict index 3, plist 2F559B98, plist index 167
24:283  0:008  OC: Prelinked blocker (Exclude) result 0 for com.apple.iokit.IOSkywalkFamily () - Success

# kexts loaded AFTER blocking IOSkywalkFamily
24:292  0:008  OCAK: as.vit9696.Lilu got KXLD 0 0
24:369  0:008  OC: Prelinked injection Kexts\Other\Lilu.kext () - Success

24:377  0:008  OCAK: org.netkas.FakeSMC got KXLD 0 0
24:459  0:009  OC: Prelinked injection Kexts\Other\FakeSMC.kext () - Success

24:468  0:008  OCAK: as.vit9696.WhateverGreen got KXLD 0 0
24:557  0:009  OC: Prelinked injection Kexts\Other\WhateverGreen.kext () - Success

24:570  0:012  OCAK: as.vit9696.AppleALC got KXLD 0 0
24:624  0:021  OC: Prelinked injection Kexts\Other\AppleALC.kext () - Success

24:633  0:008  OCAK: as.acidanthera.mieze.IntelMausi got KXLD 0 0
24:714  0:008  OC: Prelinked injection Kexts\Other\IntelMausi.kext () - Success

24:724  0:010  OCAK: com.dhinakg.USBToolBox.kext got KXLD 0 0
24:798  0:008  OC: Prelinked injection Kexts\Other\USBToolBox.kext () - Success

24:856  0:007  OCAK: org.slice.IntelCPUMonitor got KXLD 0 0
24:904  0:011  OC: Prelinked injection Kexts\Other\IntelCPUMonitor.kext () - Success

24:913  0:008  OC: Prelinked injection Kexts\Other\UTBMap.kext () - Success

24:921  0:008  OCAK: org.slice.RadeonMonitor got KXLD 0 0
24:958  0:008  OC: Prelinked injection Kexts\Other\RadeonMonitor.kext () - Success

24:970  0:011  OCAK: com.apple.iokit.IOSkywalkFamily got KXLD 0 0
25:110  0:014  OC: Prelinked injection Kexts\14\IOSkywalkFamily.kext () - Success

25:119  0:008  OCAK: com.apple.iokit.IO80211FamilyLegacy got KXLD 0 0
25:521  0:014  OC: Prelinked injection Kexts\14\IO80211FamilyLegacy.kext () - Success

25:531  0:009  OCAK: com.apple.driver.AirPort.BrcmNIC got KXLD 0 0
25:635  0:012  OC: Prelinked injection Kexts\14\IO80211FamilyLegacy.kext\Contents\PlugIns\AirPortBrcmNIC.kext () - Success

25:643  0:008  OC: Prelink size 77262848 kext offset 62218240 reserved 15044608
25:651  0:008  OCAK: KC TEXT is 65536 bytes with 62104 Mach-O headers need 960
```

b1t0y, 2023-10-23:
@SergeySlice 
> Yes, BlockSkywalk assumes minkernel=23 and Policy=Exclude.

Does this also set the IO80211FamilyLegacy.kext and its sub kext to min kernel 23? Just askin because all 3 kext needed min kernel to be 23 in OC to work. Thanks.

perez987, 2023-10-23:
@b1t0y 
I think that putting these kexts inside Kexts/14 folder they are loaded only on Sonoma. Not sure if it’s the same as min kernel.

SergeySlice, 2023-10-24:
> @SergeySlice
> 
> > Yes, BlockSkywalk assumes minkernel=23 and Policy=Exclude.
> 
> Does this also set the IO80211FamilyLegacy.kext and its sub kext to min kernel 23? Just askin because all 3 kext needed min kernel to be 23 in OC to work. Thanks.

Does it mean that if you boot Sonoma and not set min kernel the Sonoma will crash?

b1t0y, 2023-10-24:
@SergeySlice 
> Does it mean that if you boot Sonoma and not set min kernel the Sonoma will crash? Do you need to set min kernel=23.0.0.0 or just 23?

Min kernel=23.0.0

[image]

b1t0y, 2023-10-24:
> @b1t0y @SergeySlice I've tried OpenCore and Sonoma boots fine without MinKernel=23.0.0 (MinKernel=23 would be also valid) in the 3 added kexts.

Hmm. Have you tried resetting nvram on oc and see if iit still works. Im thinking that min kermel still on nvram. Coz mine did not for some reason. Will try again on OC.

perez987, 2023-10-24:
@b1t0y 
I’ll try and comment.

perez987, 2023-10-24:
@b1t0y 
Tried resetting NVRAM in the OpenCore picker, Sonoma boots fine without MinKernel in the 3 kexts.

b1t0y, 2023-10-27:
@perez987 
> @b1t0y Tried resetting NVRAM in the OpenCore picker, Sonoma boots fine without MinKernel in the 3 kexts.

Do you have any success making it to work?

perez987, 2023-10-27:
@b1t0y 
Nope. I stopped trying tests 3 days ago, without success. If @SergeySlice has anything new, I'll try it again. But so far it hasn't worked.

perez987, 2023-11-14:
@SergeySlice 
Closing the issue, when there are news it can be discussed again.

closed 2023-11-14 by perez987

perez987, 2024-01-31:
@b1t0y @Striker770 @etyel @dateno1 @jonasw8 

Clover release 5157 has fixed this issue, now you can have Broadcom wifi on Sonoma if:

- `IOSkywalk` is blocked
- `IOSkywalkFamily`, `IO80211FamilyLegacy` and `AirPortBrcmNIC` are added
- `CsrActiveConfig=0x803`

Thank you @SergeySlice and @jief666

5T33Z0, 2026-10-10:
> [@b1t0y](https://github.com/b1t0y) [@Striker770](https://github.com/Striker770) [@etyel](https://github.com/etyel) [@dateno1](https://github.com/dateno1) [@jonasw8](https://github.com/jonasw8) [...]

I think the BlockSkywalk feature is broken again. While using r5157 booting Sequoia woorks with these settings, with r5175, the system panicks instantly. So, somewhere down the road, Skywalk block broke again.

LeeBinder, 2026-10-17:
@5T33Z0 @Lzy2859905CH 

Here's the solution, verified by me:

1. open Clover /EFI/CLOVER/config.plist in a TEXT editor like TextEdit or alike
2. find `<key>KernelPm</key>` or if that key doesn't exist `<key>KernelLapic</key>`
3. after
```
<key>KernelLapic</key>
<false/>
<key>KernelPm</key>
<false/>
```
paste:
```
<key>KextsToBlock</key>
 <array>
	<dict>
		<key>Comment</key>
		<string>Allow IOSkywalk Downgrade</string>
		<key>Disabled</key>
		<false/>
		<key>MatchOS</key>
		<string>14.x,15.x,26.x</string>
		<key>Name</key>
		<string>com.apple.iokit.IOSkywalkFamily</string>
	</dict>
 </array>
```
My sections looks like this:

[image]

4. save and reboot. Your hack should boot normally again w/o KP and Wi-Fi should also work again as with OC even with Clover.

<ins>Background:</ins>

see release notes for [Clover 5168](https://github.com/CloverHackyColor/CloverBootloader/releases/tag/5168) (source only, no compile) from Feb 14 this year:

> New section KextsToBlock instead of single BlockSkywalk

meaning, with that release, @SergeySlice **deactivated** the BlockSkywalk Quirk, **replacing it** with a completely new key section..

How to use it see [config-sample.plist#L786](https://github.com/CloverHackyColor/CloverBootloader/blob/995bf1947cbc50d05dc117531ac6f5ea5e4914c2/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist#L786)

@mackie100 (Clover Configurator) did not pick up on this, either. @5T33Z0 I'm sure you'll gladly include this in Clover-Crate.. 😎

5T33Z0, 2026-10-18:
@LeeBinder Thanks. It's working now. I think what actually caused the panic was the presence of the 2 plugin kexts contained within `Airportbrcmfixup.kext`.

LeeBinder, 2026-10-18:
@5T33Z0 glad it's working for you now, too. I don't even use Airportbrcmfixup.kext but still had the kernel panic with a Skywalk error, which was solved right away after inserting that key. Your plugins related KP was one step beyond the Skywalk KP

LeeBinder, 2026-10-19:
👍 https://github.com/5T33Z0/Clover-Crate/tree/main/Kernel_And_Kext_Patches#blockskywalk