CloverHackyColor/CloverBootloader #640: Request : Add 'Block kext'

dateno1, 2023-10-10:
Please Add 'Block kext' like OC

I think It better then 'BlockSkywalk'

Example
Kext name : IOSkywalkFamily.kext (Path is not need)
Min Version : 11 (if not inputted 10.x or upper)
Max Version : 16 (if not inputted Min version~upper)
Enabled : Enabled/Disabled

It not only for 'IOSkywalkFamily.kext'

1. Can block 'IOSkywalkFamily.kext' for Broadcom WLan for Sonoma
2. Can block 'some of kext' for future update (Maybe MaxOS 15?)
3. Can block 'IO80211Family.kext' for Old WLan for new MacOS (Example : DW1520@Big Sur) (Block Internal 'IO80211Family.kext' and Add old 'IO80211Family.kext' to /efi/clover/kext

LeeBinder, 2023-12-10:
previous such requests and discussions for reference:

- https://github.com/CloverHackyColor/CloverBootloader/issues/350
- https://www.insanelymac.com/forum/topic/343301-block-injected-kexts-using-configplist/
- https://www.tonymacx86.com/threads/clover-how-to-disable-kext-to-load.203679/
- https://www.insanelymac.com/forum/topic/325619-kext-blocking-with-clover/

jief666, 2024-01-04:
Could you try this : [CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13828780/CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip) ?

LeeBinder, 2024-01-04:
Will do once I'm back tomorrow. Would be great if it was possible again (currently, to block the Skywalk kext in macOS Sonoma to revive BCM Wi-Fi not just in OC but also in Clover)

jief666, 2024-01-04:
Looks lite it works for others. If it works for you, please close this issue.

dateno1, 2024-01-04:
> Will do once I'm back tomorrow. Would be great if it was possible again (currently, to block the Skywalk kext in macOS Sonoma to revive BCM Wi-Fi not just in OC but also in Clover)
It not only for Skywalk

It for All kext

LeeBinder, 2024-01-05:
@dateno1 sure, "Darkness Angel", jief would issue a Clover binary which will block _all_ kexts, that would make _so_ much common sense .. - geez.. 👎 

.. about to test

LeeBinder, 2024-01-05:
@jief666 kextsToPatch matchos - before I bug around too much - which of these work:

1. 14.xx.x
2. 14.x
3. 14
4. 14+
5. all of the above/ doesn't matter
 
?

LeeBinder, 2024-01-05:
@jief666 : Clover GUI stalls with [CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13828780/CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip) as soon as I hit any of my macOS icons, not just Sonoma but also Monterey. Clicking onto my Windows icon correctly chainloads the Windows bootloader.

That even happens with a restored config.plist _without_ added kexttopatch entry, so it can't be that.

Naturally, I not just overwrote CloverX64.efi with yours, but also duplicated it, renamed the file clone to BOOTX64.efi and also overwrote BOOTX64.efi in EFI/boot.

To make sure there's no dependency mismatch, next I installed the latest Clover beta Clover_r5156.pkg from CircleCI (bd46eeffe), again replaced CloverX64.efi and BOOTX64.efi with your 0094e0c-jief - same thing .. :(

Needless to say, BCM kexts for Sonoma _only_ in the appropriate/ dedicated `/Volumes/EFI/EFI/CLOVER/kexts/14` folder

attached: [preboot.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13842540/preboot.log.zip)

jief666, 2024-01-05:
14 folder is for all 14.
14.x is for a specific version.
I'm not sure what name you should use. I think matchos should 23. But again not sure.

>  I don't see anybody who confirmed it's working.

There is few people on the forum who said that it works, even the skywalk block. Also an issue : https://github.com/CloverHackyColor/CloverBootloader/issues/651#issuecomment-1877018338

Anyway, if it's not working for you, well, I introduced a bug. I have another user saying the same. Let me see with him and I'll get back to you.

LeeBinder, 2024-01-05:
I'll test the `BlockSkywalk` Quirk next separately with your latest Clover beta from CircleCI.

Which forum are you referring to - can you please post a link to that topic?

So what happens if you boot with 0094e0c CloverX64.efi and BOOTX64.efi. Can you boot macOS, or do you get stuck in frozen GUI, too?

.. and @dateno1 , what about you.

> I think matchos should 23. But again not sure.

I did some more reading, it's supposed to be 14.x. So this is how the section looks at in my config.plist (but again, even without -> freeze):

```
			<dict>
				<key>Comment</key>
				<string>Block macOS Sonoma built-in IOSkywalkFamily.kext</string>
				<key>Disabled</key>
				<false/>
				<key>Name</key>
				<string>com.apple.iokit.IOSkywalkFamily</string>
				<key>Find</key>
				<data>SU9LaXQ=</data>
				<key>Replace</key>
				<data>SU9LaXM=</data>
				<key>InfoPlistPatch</key>
				<true/>
				<key>MatchOS</key>
				<string>14.x</string>
			</dict>
```

dateno1, 2024-01-06:
> @dateno1 it wouldn't make sense for jief to issue a Clover binary which would block _all_ kexts..
> 
> .. about to test
> 
> [EDIT] I now see what you mean. You need to quote and edit correctly and insert a line break before "It not only for Skywalk" so it doesn't end up in your quote of mine.
> 
> .. and you meant to say: "Procedure/ application is not only limited to Skywalk but can be applied to _any_ kext" ;)

I want to add function like 'Block some of kext from /S/L/E'

It can used for future hardware degrade

LeeBinder, 2024-01-06:
yes, @dateno1 , this is about blocking kexts from S/L/E..

The question is, does [CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13828780/CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip) work for you, following the instructions:

> don't just overwrite `CloverX64.efi`, but <ins>also **duplicate** it</ins>, rename the file you just clone to `BOOTX64.efi` and **also** overwrite `BOOTX64.efi` in `EFI/boot`!!!

and (optionally at this point because it does not influence the lock-up) adding
```
<dict>
      <key>Name</key>
      <string>com.apple.iokit.IOSkywalkFamily</string>
      <key>Find</key>
      <data>SU9LaXQ=</data>
      <key>Replace</key>
      <data>SU9LaXM=</data>
      <key>InfoPlistPatch</key>
      <true/>
      <key>MatchOS</key>
      <string>14.x</string>
      <key>Comment</key>
      <string>Block macOS Sonoma built-in IOSkywalkFamily.kext</string>
      <key>Disabled</key>
      <false/>
</dict>
```
or does your Clover GUI also freeze/ lock up.

dateno1, 2024-01-06:
> yes, @dateno1 , this is about blocking kexts from S/L/E..
> 
> The question is, does [CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13828780/CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip) work for you, following the instructions:
> 
> > don't just overwrite `CloverX64.efi`, but also **duplicate** it, rename the file you just cloned to `BOOTX64.efi` and **also** overwrite `BOOTX64.efi` in `EFI/boot`!!!
> 
> (and - optionally at this point because it doesn't influence the lock-up - add
> 
> ```
> <dict>
>       <key>Name</key>
>       <string>com.apple.iokit.IOSkywalkFamily</string>
>       <key>Find</key>
>       <data>SU9LaXQ=</data>
>       <key>Replace</key>
>       <data>SU9LaXM=</data>
>       <key>InfoPlistPatch</key>
>       <true/>
>       <key>MatchOS</key>
>       <string>14.x</string>
>       <key>Comment</key>
>       <string>Block macOS Sonoma built-in IOSkywalkFamily.kext</string>
>       <key>Disabled</key>
>       <false/>
> </dict>
> ```
> 
> to config.plist)
> 
> or does your Clover GUI also freeze/ lock up on your Hack.
Sorry my mistake

It not mean Clover GUI Freeze.

Clover GUI is OK but Can't boot SONOMA with beta version

It seem partly work (can show boot progress by before apple logo) but no apple logo and black screen

Also 'BlockSkywalk' quirk is not applied yet (I can't boot MacOSX with/without quirk&kexts)

LeeBinder, 2024-01-06:
@dateno1 would you please download and unzip this archive:

[BOOTX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13850532/BOOTX64.efi.zip)

then copy `BOOTX64.efi` to `EFI/boot`, overwriting the existing `BOOTX64.efi`, reboot into Clover, and report back if that changed anything for you?

---------------
@dateno1 이 아카이브를 다운로드하고 압축을 풀어 주시겠습니까?

[BOOTX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13850532/BOOTX64.efi.zip)

그런 다음 `BOOTX64.efi`를 `EFI/boot`에 복사하고 기존 `BOOTX64.efi`를 덮어쓴 다음 Clover로 재부팅하고 변경된 사항이 있으면 다시 보고하시겠습니까?

dateno1, 2024-01-07:
> previous such requests and discussions for reference:
> 
>     * [[NEEDED] Blocking kexts from Injection using config.plist to prevent macOS from not booting #350](https://github.com/CloverHackyColor/CloverBootloader/issues/350)
> 
>     * https://www.insanelymac.com/forum/topic/343301-block-injected-kexts-using-configplist/
> 
>     * https://www.tonymacx86.com/threads/clover-how-to-disable-kext-to-load.203679/ (locked topic)
> 
>     * https://www.insanelymac.com/forum/topic/325619-kext-blocking-with-clover/

I want to function like this but  clover is not changed in few years

I want to request to add new function like them

> @dateno1 would you please download and unzip this archive:
>
> [BOOTX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13850532/BOOTX64.efi.zip)
>
> then copy `BOOTX64.efi` to [...]

I think you are not understand what is problem

skywalk is not problem (It already fixed on beta version and work well)

That not fix or prevent other problems on future update

Also lastest beta version is better than 240104 version

jief666, 2024-01-07:
@dateno1 From what I understand, you'd like a way to block any kext you'd like. I thought the feature was already there ! But we can do it. Can you open another issue requesting that feature ?
@LeeBinder Lets' summurize : the issue about blocking skywalk is resolved, but a new issue arised : the graphic freeze. Am I right. If yes, we'll close this one and open a new one.

LeeBinder, 2024-01-07:
dateno1, I sure DO understand what the "problem" is, to enable a general  'block kext' function to prevent what you call 'future hardware degrade' - it was me who posted the 4 links you quoted.

I'm also aware that jief666 has fixed the BlockSkywalk quirk in current beta versions, and that it works well in that limited regard.

At this stage, we can (actually should!) use the Skywalk kext as a TEST to see which effects jief666's edits have - THAT's where we're at.

So would you PLEASE, for heaven's sake, do the following:

1. deactivate the BlockSkywalk quirk in config.plist

2. add the patch according to [the instructions](https://github.com/CloverHackyColor/CloverBootloader/issues/640#issuecomment-1879216632)

3. in addition to [CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13828780/CloverX64-2024-01-04-10-59-02-0094e0c-jief.zip), also apply [BOOTX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13850532/BOOTX64.efi.zip)

reboot and share the result with us.

Thank you

LeeBinder, 2024-01-07:
> @dateno1 From what I understand, you'd like a way to block any kext you'd like. I thought the feature was already there !

@jief666 see https://github.com/CloverHackyColor/CloverBootloader/issues/350#issuecomment-770341556. It was indeed working before r5122, but then came the switch to OC code base which must have nuked this feature. I thought you were aware of this.

> Can you open another issue requesting that feature ?

jief666 please no: @dateno1's request was and is clear both in the title and in his first post in this issue topic, that it's a general request, separate from the other BlockSKywalk quirk topic. Also we've already had other request issues here regarding this, so this issue here should be enough, and hopefully the last in this regard

> Lets' summurize : the issue about blocking skywalk is resolved, but a new issue arised : the graphic freeze. Am I right.

Again, not exactly: restoring the BlockSkywalk quirk which was introduced in 5155 and broken in 5156 was solved as per the other issue, but this is to restore the regression that general blocking of kext in S/L/E - which was working before - has not been working anymore since adaptation of OC code, as noted before by @stevezhengshiqi 

If you refer to Clover GUI lock-up after hitting an icon which is supposed to boot any macOS as 'graphic freeze': I cannot test if the feature requested in this (and previously other) topics might actually be working with your Clover64/Boot64.efi because of the lock-up. So the freeze is in the way of the testing and needs to be resolved first as part of this debugging.

Hope all is clear now so you can proceed ;)

LeeBinder, 2024-01-07:
As a matter of fact, you already have the working code in the function behind the BlockSKywalk quirk:

1. blocks a kext with a specified name or identifier in S/LE
2. only blocks if for a specific version of macOS, in this case Sonoma 14.x or kernel 23.x

So in a nutshell, that code would need to be linked to the `Kernel and Kext Patches`/ `KextsToPatch` section and executed with the parameters from there. Maybe you can even call the same function, or duplicate that function into one which accepts parameters.

dateno1, 2024-01-07:
> @dateno1 From what I understand, you'd like a way to block any kext you'd like. I thought the feature was already there ! But we can do it. Can you open another issue requesting that feature ? [...]

Few years ago some of people request this function but nothing changed and Now apple change kext&make problem

I was worry apple create 'anther skywalk' again

> dateno1, I sure DO understand what the "problem" is, to enable a general 'block kext' function to prevent what you call 'future hardware degrade' - it was me who posted the 4 links you quoted.
>
> I'm [...]

I was tested BlockSkywalk quirk on 3 boards
GA-H110N, GA-Z390M : Work well
GA-B365M Aorus Elite : Can't boot MacOSX with Clover beta version (with/without BlockSkywalk quirk)

> As a matter of fact, you already have the working code in the function behind the BlockSKywalk quirk:
>
>     1. blocks a kext with a specified name or identifier in S/LE
>
>     2. only blocks if for a [...]

I don't think it duplicated

BlockSkywalk is just function only for Skywalk

It not used for another kexts on /S/L/E

LeeBinder, 2024-01-08:
Once more, this is _not_ the "BlockSkywalk quirk" issue topic..

.. and neither is this a "_Can't boot MacOSX on GA-B365M Aorus Elite with Clover beta version_" issue topic - that's completely unrelated (and if "_with/without BlockSkywalk quirk_" matters even less)...

Sure, "_BlockSkywalk is function only for Skywalk_", but what can be applied to kext a in S/L/E for macOS 14 can be applied to kext b in S/L/E to macOS xy.z, too, can't it? That's what functions and algorithms are about....

dateno1, since there's obviously at least a dire language barrier, I highly recommend to keep quiet now since all has been said and nothing productive can be added anymore. Let's jief666 give the space to fix this regression introduced with r5123 and re-add it to the current OC code based Clover.

Peace out for now

dateno1, 2024-01-08:
> Once more, this is _not_ the "BlockSkywalk quirk" issue topic..
> 
> .. and neither is this a "_Can't boot MacOSX on GA-B365M Aorus Elite with Clover beta version_" issue topic - that's completely unrelated (and if "_with/without BlockSkywalk quirk_" matters even less)...
> 
> Sure, "_BlockSkywalk is function only for Skywalk_", but what can be applied to kext a in S/L/E for macOS 14 can be applied to kext b in S/L/E to macOS xy.z, too, can't it? That's what functions and algorithms are about....
> 
> dateno1, since there's obviously at least a dire language barrier, I highly recommend to keep quiet now since all has been said and nothing productive can be added anymore. Let's jief666 give the space to fix this regression introduced with r5123 and re-add it to the current OC code based Clover.
> 
> Peace out for now
> 
> 다시 한 번 말씀드리지만, 이것은 "BlockSkywalk 별난" 문제 주제가 _아닙니다_.
> 
> .. 그리고 이것은 "_Clover 베타 버전이 포함된 GA-B365M Aorus Elite에서 MacOSX를 부팅할 수 없습니다_" 문제 주제도 아닙니다. 이는 완전히 관련이 없습니다(그리고 "_BlockSkywalk 기발함이 있거나 없는_"이 훨씬 덜 중요하다면)...
> 
> 물론, "_BlockSkywalk는 Skywalk_ 전용 기능입니다." 하지만 macOS 14용 S/L/E의 kext a에 적용할 수 있는 것은 S/L/E의 kext b에도 적용할 수 있고 macOS xy.z에도 적용할 수 있습니다. 그렇지? 그것이 바로 함수와 알고리즘에 관한 것입니다....
> 
> dateno1, 적어도 심각한 언어 장벽이 분명히 있기 때문에 모든 것이 말해지고 더 이상 생산적인 내용을 추가할 수 없으므로 지금은 조용히 있는 것이 좋습니다. jief666은 r5123에 도입된 이 회귀를 수정하고 이를 현재 OC 코드 기반 Clover에 다시 추가할 공간을 제공합니다.
> 
> 지금은 평화를 누리세요

It not only for sonoma

If someone add this function at 2022 'BlockSkywalk' and some of quirks on future is not needed

closed 2024-01-08 by dateno1

LeeBinder, 2024-01-09:
2022? LOL we're already in 2024..

You're painfully confused, mate. Hope you can get some help [image]

dateno1, 2024-01-09:
> 2022? LOL we're already in 2024..
> 
> You're painfully confused, mate. Hope you can get some help ![42545-ambulance-icon](https://private-user-images.githubusercontent.com/39203497/295152488-a6baad0c-ba0b-4a12-b9b2-b1ae52916774.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4NDI5MzMsIm5iZiI6MTcwNDg0MjYzMywicGF0aCI6Ii8zOTIwMzQ5Ny8yOTUxNTI0ODgtYTZiYWFkMGMtYmEwYi00YTEyLWI5YjItYjFhZTUyOTE2Nzc0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTA5VDIzMjM1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0ZmUyMTY0ODA5NWQ3YjdiNzM4MGQyNmJkNTQ3ZTExMDE3Y2U1OGEwZmY2MGM5NTg1NDY0NjA4NDQ4NjQ4NjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.78mVtbAWPIZi9OWVe6WwZaXDbDTJ5cKBdAiryUPrKmA)

> previous such requests and discussions for reference:
>
>     * [[NEEDED] Blocking kexts from Injection using config.plist to prevent macOS from not booting [...]

It duplicated request

It already said at past but no one do that

I think it too late

LeeBinder, 2026-10-17:
This feature exists ever since Clover 5168/5169, see [my comment here](https://github.com/CloverHackyColor/CloverBootloader/issues/634#issuecomment-6242012425.).