CloverHackyColor/CloverBootloader #641: The latest Clover is not working in the Dell Inspiron 1520 Legacy BIOS.

hidem5, 2023-10-15:
Specifically, the number 6 appears on the left edge of the screen and then freezes with a black screen.
The version of Clover that works on this machine is up to 5100.

Are there any special operations or tasks that need to be done on this type of machine in order to get 5100 or later to work?

SergeySlice, 2023-11-16:
I had Dell Inspiron 1520 several years ago and it was booted by Clover. So it is possible.

closed 2023-11-16 by SergeySlice

hidem5, 2023-11-17:
The last version that worked was wrong.
5107 to be exact.

I then tested a new install of the latest 5156 on a USB and it did not boot.

hidem5, 2023-11-23:
5107:
10.4 and 10.5 will not start if kextcache is present.
It does not start with -f either.
couldn't alloc AppleALC,SMCBatterymanager,etc

I wanted a new Clover to solve this problem.

5112:
Only the Clover GUI startup was tested.

5123.1:
10.5 will not boot, even after deleting kextcache. couldn't alloc Class efi kexts is displayed.
10.4 and 10.6 also do not boot unless I select inject kext from the boot menu options. kext in EFI does not seem to be loaded automatically.
10.13 boots with no problem.

FuzzyMatch is set to Auto.
The same was true when it was deactivated.

5132:
Clover GUI shows up, but 10.4, 10.5, and 10.6 show errors in red text on a black screen.

5155
5156:
Nothing is displayed after the number 6 blinks.no Clover GUI

If an older Clover was installed on the internal SSD, the older Clover GUI will appear after the 6 blinks.
However, when I try to boot, I get a black screen with a red error message.