CloverHackyColor/CloverBootloader #643: Unable To Build Clover With Secure Boot (Linux & Windows)

WyndRyder, 2023-10-28:
I'm trying build Clover on my Ubuntu system as I'm trying to multi-boot between Windows and Linux distros while Secure Boot is enabled (I need it for certain Windows functionality). While doing this in Ubuntu, I've encountered the following:

**BuildMe Menu Here**

Please enter your choice: 1
WORKSPACE: /home/aelyth/src/CloverBootloader
EDK_TOOLS_PATH: /home/aelyth/src/CloverBootloader/BaseTools
CONF_PATH: /home/aelyth/src/CloverBootloader/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /home/aelyth/src/CloverBootloader/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /home/aelyth/src/CloverBootloader/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /home/aelyth/src/CloverBootloader/Conf/target.txt
[BUILD CLOVER]
TOOLCHAIN_DIR: /usr
NASM_PREFIX: /usr/bin/
NASM_VER: 2.16.01
Initializing workspace
recreate Conf folder
WORKSPACE: /home/aelyth/src/CloverBootloader
EDK_TOOLS_PATH: /home/aelyth/src/CloverBootloader/BaseTools
CONF_PATH: /home/aelyth/src/CloverBootloader/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /home/aelyth/src/CloverBootloader/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /home/aelyth/src/CloverBootloader/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /home/aelyth/src/CloverBootloader/Conf/target.txt
Building tools as they are not found
make: Entering directory '/home/aelyth/src/CloverBootloader/BaseTools'
make -C Source/C
make[1]: Entering directory '/home/aelyth/src/CloverBootloader/BaseTools/Source/C'
Attempting to detect HOST_ARCH from 'uname -m': x86_64
Detected HOST_ARCH of X64 using uname.
mkdir -p .
make -C Common
make[2]: Entering directory '/home/aelyth/src/CloverBootloader/BaseTools/Source/C/Common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/aelyth/src/CloverBootloader/BaseTools/Source/C/Common'
make -C BrotliCompress
make[2]: Entering directory '/home/aelyth/src/CloverBootloader/BaseTools/Source/C/BrotliCompress'
gcc  -c -I ./include -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/  -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  dec/decode.c -o dec/decode.o
dec/decode.c:1945:41: error: argument 2 of type 'const uint8_t *' {aka 'const unsigned char *'} declared as a pointer [-Werror=vla-parameter]
 1945 |     size_t encoded_size, const uint8_t* encoded_buffer, size_t* decoded_size,
      |                          ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from dec/decode.c:7:
./include/brotli/decode.h:204:19: note: previously declared as a variable length array 'const uint8_t[encoded_size]' {aka 'const unsigned char[encoded_size]'}
  204 |     const uint8_t encoded_buffer[BROTLI_ARRAY_PARAM(encoded_size)],
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dec/decode.c:1946:14: error: argument 4 of type 'uint8_t *' {aka 'unsigned char *'} declared as a pointer [-Werror=vla-parameter]
 1946 |     uint8_t* decoded_buffer) {
      |     ~~~~~~~~~^~~~~~~~~~~~~~
./include/brotli/decode.h:206:13: note: previously declared as a variable length array 'uint8_t[*decoded_size]' {aka 'unsigned char[*decoded_size]'}
  206 |     uint8_t decoded_buffer[BROTLI_ARRAY_PARAM(*decoded_size)]);
      |     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [../Makefiles/footer.makefile:21: dec/decode.o] Error 1
make[2]: Leaving directory '/home/aelyth/src/CloverBootloader/BaseTools/Source/C/BrotliCompress'
make[1]: *** [GNUmakefile:76: BrotliCompress] Error 2
make[1]: Leaving directory '/home/aelyth/src/CloverBootloader/BaseTools/Source/C'
make: *** [GNUmakefile:19: Source/C] Error 2
make: Leaving directory '/home/aelyth/src/CloverBootloader/BaseTools'

**I'm noticing the compiler is treating all warnings as errors (-Werror flag). I'm wondering if it has to do with the usage of VLAs as function parameters.**

I'll be the first to admit that this is my first time doing something like this. But I'm very interested in learning and I think I'm close (**have downloaded the files from Microsoft CA for signing the binaries)**. 

Any help is greatly appreciated!

SergeySlice, 2023-10-30:
make -C Source/C has a problem? It's impossible! Reload all.

SergeySlice, 2023-10-30:
I made it in Sonoma 14.0 with Xcode15beta without problems.

WyndRyder, 2023-10-30:
The compiler flags the use of **const uint8_t* encoded_buffer** and **uint8_t* decoded_buffer** as pointers and notes that they were previously declared as variable-length arrays. 
Are we sure the pointers should be used in this manner??!?

WyndRyder, 2023-11-05:
Still no luck. Is there a way to build without the -Werror option? I've been digging around in the build scripts, but can't seem to find it.

SergeySlice, 2023-11-16:
Any movements?

ryanrudolfoba, 2024-01-05:
i'm trying to build this on my CentOS 9 machine and getting similar error. do i need a Mac to build this or any Linux flavor is fine to build?

[image]

[image]

druidgeek, 2024-01-05:
Yeah, same error here for me as well. I just gave up on it...

SergeySlice, 2024-01-13:
Compilation errors were resolved. Take sources of 5157.
But the SecureBoot as technology is not finished developing. Be the developer for this.

ryanrudolfoba, 2024-01-13:
> Compilation errors were resolved. Take sources of 5157. But the SecureBoot as technology is not finished developing. Be the developer for this.

Thanks for getting back. I'm still getting build errors -
[image]

However it looks like its the EDK BaseTools that is giving me the issue and not Clover related. I've compiled the BaseTools with the -Wno-error flag and that did the trick. I'm able to proceed to build Clover. thank you!

druidgeek, 2024-01-14:
Care to share the love? What did you run (flags and all) to compile the BaseTools? Please and thanks...

maxpicelli, 2024-03-19:
Oi pessoal, estou aprendendo a compilar no Linux, e é possível adicionar algum comando para concluir a compilação como esta -Wno-error? If some could help me to know how i can fix that, i net same error like image above. Thank you!

maxpicelli, 2024-03-19:
When I try compile on macOs I get this error. Sorry for this issues questions. 

[image]

maxpicelli, 2024-03-19:
[image]

SergeySlice, 2024-03-31:
There is the message "...check logs in...". Follow please!

maxpicelli, 2024-03-31:
Thanks @SergeySlice  im think now im fix with your command, sorry about log, im lost or delete. But im think now is fix with your lesson:

and im using this command to install GCC 

`mkdir -p $HOME/src && cd $HOME/src
git clone -q https://github.com/CloverHackyColor/CloverBootloader
cd CloverBootloader && git submodule update --init --recursive
cd OpenCorePkg && git checkout master && git pull
cd ..
./build_gcc13.sh
 `

[image]

closed 2024-06-03 by SergeySlice