CloverHackyColor/CloverBootloader #647: Redundant boot devices, try all until successful

fiesh, 2023-11-02:
I'm trying to boot from NVMe on one of these awful Dell servers that fully recognize and support the NVMe devices yet block booting from them so force you into Dell's vendor lock.

There are 8 NVMe disks installed, and for redundancy purposes they all have the same EFI boot partition.  So I'd like to make Clover boot to the first one that works, thus supporting device failure.

Alas I cannot figure out a way to do that.  I'm fine with specifying the UUIDs, though I'd prefer to not even have to do that so devices can be switched without the read-only memory stick having to change.  Booting from one hard-coded UUID works fine.

Feedback would be greatly appreciated.

SergeySlice, 2023-11-16:
It is your special configuration and nobody else having same one. So you should be a developer to make that you want.
If you have special question then ask.

fiesh, 2023-11-16:
> It is your special configuration and nobody else having same one. So you should be a developer to make that you want. If you have special question then ask.

The question is simply: How do I specify several boot devices and have Clover try them in order until one works?

SergeySlice, 2023-12-02:
There is no such functionality as 15 years was not asked.
The boot sequence is unique until Clover GUI. Then you can choose what system to boot.

jief666, 2024-01-04:
@fiesh What does mean "boot one that works" ? What is "not working" for a boot ? A crash ? Or a SSD that is not there anymore ?

Clover can automatically boot the last booted partition. If that SSD disappear, Clover stops at the menu.
We can imagine a feature, especially for you, that boot the first available partition when the last booted partition is not there. But is the first available always the right one ?

fiesh, 2024-01-04:
@jief666 _not works_ is meant to mean _is not bootable_.  This is purely meant as a form of redundancy against device failure, so all possible boot options lead to the same system being booted.  So basically any boot option is equally good.  Since I intend to set the USB device to read-only by a hardware switch, saving the _last booted partition_ wouldn't work as such anyway.

Thank you!

jief666, 2024-01-04:
Which device can fail, the one that contains Clover, or the one that contains a bootable os?

fiesh, 2024-01-04:
> Which device can fail, the one that contains Clover, or the one that contains a bootable os?

The bootable OS partitions -- the Clover stick is assumed to work, which is why I want it on read-only to improve its longevity, especially on cheap commodity hardware like memory sticks.

jief666, 2024-01-05:
This experimental version boot the first available volume with a timeout of 10s :[CloverX64-2024-01-05-15-51-02-3192950-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13843371/CloverX64-2024-01-05-15-51-02-3192950-dirty-jief.zip).
Could you try that and tell if it fits your need ?

fiesh, 2024-01-05:
> This experimental version boot the first available volume with a timeout of 10s [...]

Awesome, thank you!  Will be at the server in question again next week and report back!

fiesh, 2024-01-08:
I have a hard time getting it to boot, @jief666 

Am I supposed to configure something special?  I've tried to

* Use the working `config.plist` but remove the `Boot` key/value pair from it
* Generate a new `config.plist` by selecting Clover's option to write all bootable devices to the file
* Removing `config.plist` all together.

jief666, 2024-01-09:
The version I'm sending you has a quick experimental hack that take precedence over what's in config.plist.
10 sec timeout and first entry to boot. So put back your original config.plist. You shouldn't need to modify anything in it for now. Clover should find all the bootable devices. This list is NOT in config.plist anyway.
However, I found a nasty memory bug that made clover erratic.

So let's try again with this version : [CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13875638/CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.zip)

jief666, 2024-01-13:
> I tried again but to no avail

You'll have to take the time to describe better what does it do. I can't see your screen, you're not sending any log.
Please first, you have to activate debug in boot section of config.plist (see forum if needed), second, tell me what it does instead of just "not working".

The version I sent is supposed to boot the first volume, meaning the icon on the left. Whatever it's a custom entry on not.

> So put back your original config.plist. You shouldn't need to modify anything in it for now.

Looks like you are going a bit too fast, not reading my message properly. Please slow down, do exactly what I say, not more not less, so I know what's going on. Each test, report back with the log. That's the only way I can figure out what is not working.
No, don't modify your config.plist

fiesh, 2024-01-15:
> You'll have to take the time to describe better what does it do. I can't see your screen, you're not sending any log. Please first, you have to activate debug in boot section of config.plist (see [...]

Ok sorry for the confusion, I'll explain my situation:

* I have created, for the stable version of Clover, a file `config.plist` (the one I pasted) that boots one of the volumes, which works but of course provides no redundancy.  Because of its `Timeout` of `0` and its `DefaultVolume` being set, I figured I couldn't just leave it since it boots anyway.
* This is why I used it but removed the `DefaultVolume`.
* It not working meant it just boots to Clover but stays in Clover's GUI.
* So should I leave this `config.plist` as it is (as above without `DefaultVolume`), or should I use my actual working one with the `DefaultVolume` being set?

Please clarify, then I'll boot it and provide the log contents.  Thank you!

jief666, 2024-01-15:
In the experimental version I sent you, DefaultVolume is ignored.

> you have to activate debug in boot section of config.plist (see forum if needed)

Send me a log after each test. I cannot see what's going otherwise.

fiesh, 2024-01-16:
Thanks for the explanation regarding `DefaultVolume`.  So here's my debug log:

```
0:100  0:100  MemLog inited, TSC freq: 2094951223
0:100  0:000  CPU was calibrated with ACPI PM Timer
0:100  0:000  CloverX64 : Image base = 0x6147F000
0:100  0:000  Clover ImageHandle = 622c4798
0:101  0:001  SelfDirPath = \EFI\CLOVER
0:113  0:012  === [ Starting Clover ] =========================
0:113  0:000  Now is 16.01.2024,  08:08:34 (GMT)
0:113  0:000  Starting Clover revision: 5156 (master, commit 187400d0a) on Dell Inc. EFI
0:113  0:000  Build id: 20231111155702-187400d-5156-dirty
0:113  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG -t GCC131 | --cmd-len=50000 -D NO_GRUB_DRIVERS_EMBEDDED -D LESS_DEBUG --conf=/Users/sergey/src/CloverBootloader/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC131 -n 5 | OS: 13.6.1]
0:135  0:021  --> : Firmware has 268293997 free pages (382829 in lower 4 GB)
0:135  0:000  === [ Get Smbios ] ==============================
0:135  0:000  Type 16 Index = 0
0:135  0:000  Total Memory Slots Count is adjusted from 96 to 24
0:135  0:000  Type 17 Index = 0
0:135  0:000  SmbiosTable.Type17->SlotIndex = 0
0:135  0:000  SmbiosTable.Type17->ModuleSize = 32768MB
0:135  0:000  SmbiosTable.Type17->Type = 0
0:135  0:000  SmbiosTable.Type17->Frequency = 2400MHz
0:135  0:000  SmbiosTable.Type17->Bank/Device =  A1
0:135  0:000  SmbiosTable.Type17->Vendor = 00CE063200CE
0:135  0:000  SmbiosTable.Type17->SerialNo = 334CF9D9
0:135  0:000  SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
0:135  0:000  Type 17 Index = 1
0:135  0:000  SmbiosTable.Type17->SlotIndex = 1
0:135  0:000  SmbiosTable.Type17->ModuleSize = 32768MB
0:135  0:000  SmbiosTable.Type17->Type = 0
0:135  0:000  SmbiosTable.Type17->Frequency = 2400MHz
0:135  0:000  SmbiosTable.Type17->Bank/Device =  A2
0:135  0:000  SmbiosTable.Type17->Vendor = 00CE063200CE
0:135  0:000  SmbiosTable.Type17->SerialNo = 32EC862A
0:135  0:000  SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
0:135  0:000  Type 17 Index = 2
0:135  0:000  SmbiosTable.Type17->SlotIndex = 2
0:135  0:000  SmbiosTable.Type17->ModuleSize = 32768MB
0:135  0:000  SmbiosTable.Type17->Type = 0
0:135  0:000  SmbiosTable.Type17->Frequency = 2400MHz
0:135  0:000  SmbiosTable.Type17->Bank/Device =  A3
0:135  0:000  SmbiosTable.Type17->Vendor = 00CE00B300CE
0:135  0:000  SmbiosTable.Type17->SerialNo = 199B4A3F
0:135  0:000  SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
0:135  0:000  Type 17 Index = 3
0:135  0:000  SmbiosTable.Type17->SlotIndex = 3
0:135  0:000  SmbiosTable.Type17->ModuleSize = 32768MB
0:135  0:000  SmbiosTable.Type17->Type = 0
0:135  0:000  SmbiosTable.Type17->Frequency = 2400MHz
0:135  0:000  SmbiosTable.Type17->Bank/Device =  A4
0:135  0:000  SmbiosTable.Type17->Vendor = 00CE063200CE
0:135  0:000  SmbiosTable.Type17->SerialNo = 32EC8981
0:135  0:000  SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
0:135  0:000  Type 17 Index = 4
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 5
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 6
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 7
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 8
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 9
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 10
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 11
0:135  0:000  Ignoring insane frequency value 0MHz
0:135  0:000  Type 17 Index = 12
[... 409 lines cut ...]
127:322  0:035  - [02]: 'BOOT'
127:398  0:075        AddLoaderEntry for Volume Name=BOOT, idx=0
127:434  0:036        OSVersion= 
127:517  0:082        IconName=unknown comma=-1 size=7
127:552  0:035        Full=unknown
127:587  0:035        Loader entry created for 'PciRoot(0x0)\Pci(0x1A,0x0)\USB(0x0,0x0)\USB(0x1,0x0)\USB(0x0,0x0)\HD(1,GPT,A702E595-5CA5-4A72-A689-88DB4EF34F72,0x800,0xEEF000)\EFI\BOOT\BOOTX64.efi'
127:623  0:035        Menu entry added at index 0
127:661  0:037       hiding entry because DiskKind is DISK_KIND_EXTERNAL: \EFI\BOOT\BOOTX64.efi
127:697  0:036  Entries list before ordering
127:732  0:035      Entry 0 : Boot UEFI external from BOOT (hidden)
127:768  0:035  Entries list after ordering
127:803  0:035    Entry 0 : Boot UEFI external from BOOT (hidden)
127:838  0:035  Scanning legacy ...
127:893  0:054  === [ AddCustomTool ] ===========================
127:928  0:035  found tool \EFI\CLOVER\tools\Shell64U.efi
127:964  0:036  Checking EFI partition Volume 2 for Clover
128:000  0:035   Found Clover
128:036  0:035  === [ GetEfiBootDeviceFromNvram ] ===============
128:124  0:088   - [!] efi-boot-device-data not found
128:160  0:036  Default boot entry not found
128:216  0:055  --> Before RunMainMenu: Firmware has 268287531 free pages (376363 in lower 4 GB)
128:251  0:035  SetScreenResolution: 1680x1050 - already set
```

I suppose the messages `skipped because media is ignored` are what's causing the system not to boot?

jief666, 2024-01-16:
You didn't start the last Clover efi I've sent you.
At the beginning of the log, you should see 
```
Starting Clover revision: 5157 (master, commit 8697e9380) on Dell Inc. EFI
Build id: 2024-01-15-19-21-57-8697e93-dirty-jief

```
and you have

```
0:113  0:000  Starting Clover revision: 5156 (master, commit 187400d0a) on Dell Inc. EFI
0:113  0:000  Build id: 20231111155702-187400d-5156-dirty
```
So please boot the right one and send me the log.

fiesh, 2024-01-16:
This is the version from your zip file, I think you packaged the wrong one:

```zsh
fiesh@ltsp164 /tmp % unzip CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.zip
Archive:  CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.zip
  inflating: CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.efi  
fiesh@ltsp164 /tmp % strings CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.efi|grep 'Clover revision'
Clover revision: 5156 (master, commit dcdf3e972)
```

jief666, 2024-01-16:
Sorry, previous message is wrong.
You should see :
```
Starting Clover revision: 5156 (master, commit dcdf3e972) on Dell Inc. EFI ( ->exactly as your strings cmd shows
Build id: 2024-01-09-16-20-31-dcdf3e9-dirty-jief
```
But ou have

```
0:113  0:000  Starting Clover revision: 5156 (master, commit 187400d0a) on Dell Inc. EFI (->wrong commit)
0:113  0:000  Build id: 20231111155702-187400d-5156-dirty (->wrong buildid)
```

jief666, 2024-01-16:
You can double check build id with strings CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.efi | grep "CloverBuildIdGrepTag"

fiesh, 2024-01-16:
This is what my USB boot stick looks like:

```
fiesh@ltsp164 /tmp % ls -l /mnt/efi/clover/
total 4388
-rwxr-xr-x 1 root root 2832064 Jan 11 09:00 cloverx64.efi
-rwxr-xr-x 1 root root 1598144 Jan  8 11:10 cloverx64.old
-rwxr-xr-x 1 root root   32010 Jan  8 11:10 config.old
-rwxr-xr-x 1 root root    3631 Jan 16 08:34 config.plist
drwxr-xr-x 5 root root    4096 Jan  8 11:10 drivers
drwxr-xr-x 3 root root    4096 Jan  8 11:10 kexts
drwxr-xr-x 2 root root    4096 Jan 16 08:10 misc
drwxr-xr-x 2 root root    4096 Jan  8 11:10 themes
drwxr-xr-x 2 root root    4096 Jan  8 11:10 tools
fiesh@ltsp164 /tmp % cmp /mnt/efi/clover/cloverx64.efi CloverX64-2024-01-09-16-20-31-dcdf3e9-dirty-jief.efi
fiesh@ltsp164 /tmp % 
```

Am I doing something wrong?

jief666, 2024-01-16:
Most likely ! :-)
Maybe you are booting on a complete different Clover.
I would advise using BootloaderChooser. It'll give you a text menu that will allow you to choose on which efi from which folder you'd like to boot.

jief666, 2024-01-16:
So you should consider BootloaderChooser once more if your bios can boot a file that's not called [something].efi !

jief666, 2024-01-16:
> Hmm ok, grepping for `20231111155702` shows that `cloverx64.old` matches... so I guess somehow the `.old` file was picked up. Didn't expect that :-( Sorry, will delete and try again. Sorry for the stupid mistake, this got me good, thought the `.efi` file extension was somehow honored...

That's surprising. Are you sure not to have another CLOVER folder somewhere on a hard drive ?

fiesh, 2024-01-16:
Ok, so now the timer running down from 10 seconds to 0 appears, after which the system goes to the EFI console instead of booting.  The log is this:

```
MemLog inited, TSC freq: 2094950603
CPU was calibrated with ACPI PM Timer
CloverX64 : Image base = 0x61225000
Clover ImageHandle = 622c4798
SelfVolumeRootDir = 1596909856
efiFileName=BOOTX64.EFI
SelfDirPath = \EFI\CLOVER
CloverDir = 1596910368
Self DevicePath()=PciRoot(0x0)\Pci(0x1A,0x0)\USB(0x0,0x0)\USB(0x1,0x0)\USB(0x0,0x0)\HD(1,GPT,A702E595-5CA5-4A72-A689-88DB4EF34F72,0x800,0xEEF000)\EFI\CLOVER\BOOTX64.EFI @5F7B1998
SelfDirPath = \EFI\CLOVER
=== [ Starting Clover ] =========================
Now is 16.01.2024,  10:05:03 (GMT)
Starting Clover revision: 5156 (master, commit dcdf3e972) on Dell Inc. EFI
Build id: 2024-01-09-16-20-31-dcdf3e9-dirty-jief
Build with: [jief]
--> : Firmware has 268293732 free pages (382564 in lower 4 GB)
=== [ Get Smbios ] ==============================
Type 16 Index = 0
Total Memory Slots Count is adjusted from 96 to 24
Type 17 Index = 0
SmbiosTable.Type17->SlotIndex = 0
SmbiosTable.Type17->ModuleSize = 32768MB
SmbiosTable.Type17->Type = 0
SmbiosTable.Type17->Frequency = 2400MHz
SmbiosTable.Type17->Bank/Device =  A1
SmbiosTable.Type17->Vendor = 00CE063200CE
SmbiosTable.Type17->SerialNo = 334CF9D9
SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
Type 17 Index = 1
SmbiosTable.Type17->SlotIndex = 1
SmbiosTable.Type17->ModuleSize = 32768MB
SmbiosTable.Type17->Type = 0
SmbiosTable.Type17->Frequency = 2400MHz
SmbiosTable.Type17->Bank/Device =  A2
SmbiosTable.Type17->Vendor = 00CE063200CE
SmbiosTable.Type17->SerialNo = 32EC862A
SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
Type 17 Index = 2
SmbiosTable.Type17->SlotIndex = 2
SmbiosTable.Type17->ModuleSize = 32768MB
SmbiosTable.Type17->Type = 0
SmbiosTable.Type17->Frequency = 2400MHz
SmbiosTable.Type17->Bank/Device =  A3
SmbiosTable.Type17->Vendor = 00CE00B300CE
SmbiosTable.Type17->SerialNo = 199B4A3F
SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
Type 17 Index = 3
SmbiosTable.Type17->SlotIndex = 3
SmbiosTable.Type17->ModuleSize = 32768MB
SmbiosTable.Type17->Type = 0
SmbiosTable.Type17->Frequency = 2400MHz
SmbiosTable.Type17->Bank/Device =  A4
SmbiosTable.Type17->Vendor = 00CE063200CE
SmbiosTable.Type17->SerialNo = 32EC8981
SmbiosTable.Type17->PartNo = M393A4K40BB1-CRC
Type 17 Index = 4
Type 17 Index = 5
Type 17 Index = 6
Type 17 Index = 7
Type 17 Index = 8
Type 17 Index = 9
Type 17 Index = 10
Type 17 Index = 11
Type 17 Index = 12
SmbiosTable.Type17->SlotIndex = 12
SmbiosTable.Type17->ModuleSize = 32768MB
SmbiosTable.Type17->Type = 0
[... 539 lines cut ...]
    Checking volume "BOOT" (PciRoot(0x0)\Pci(0x1A,0x0)\USB(0x0,0x0)\USB(0x1,0x0)\USB(0x0,0x0)\HD(1,GPT,A702E595-5CA5-4A72-A689-88DB4EF34F72,0x800,0xEEF000)) ... skipped because media is ignored
=== [ ScanLoader ] ==============================
- [02]: 'BOOT'
      AddLoaderEntry for Volume Name=BOOT, idx=0
      OSVersion= 
      IconName=unknown comma=-1 size=7
      Full=unknown
      Loader entry created for 'PciRoot(0x0)\Pci(0x1A,0x0)\USB(0x0,0x0)\USB(0x1,0x0)\USB(0x0,0x0)\HD(1,GPT,A702E595-5CA5-4A72-A689-88DB4EF34F72,0x800,0xEEF000)\EFI\BOOT\BOOTX64.efi'
      Menu entry added at index 0
     hiding entry because DiskKind is DISK_KIND_EXTERNAL: \EFI\BOOT\BOOTX64.efi
Entries list before ordering
    Entry 0 : Boot UEFI external from BOOT (hidden)
Entries list after ordering
  Entry 0 : Boot UEFI external from BOOT (hidden)
Scanning legacy ...
=== [ AddCustomTool ] ===========================
found tool \EFI\CLOVER\tools\Shell64U.efi
Checking EFI partition Volume 2 for Clover
 Found Clover
=== [ GetEfiBootDeviceFromNvram ] ===============
 - [!] efi-boot-device-data not found
Default boot entry not found
--> Before RunMainMenu: Firmware has 268287280 free pages (376112 in lower 4 GB)
SetScreenResolution: 1680x1050 - already set
```

jief666, 2024-01-16:
Please do not copy paste the log, but drag and drop the file instead.
I realize the "skipped because media is ignored" and then I saw :

> I suppose the messages skipped because media is ignored are what's causing the system not to boot?

Sorry, I didn't see that message before.
Yes, if you don't have all your icons on the screen, Clover cannot boot them.

Try this : [CloverX64-2024-01-16-12-09-29-8697e93-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13949060/CloverX64-2024-01-16-12-09-29-8697e93-dirty-jief.zip), the log will tell me what's wrong.

jief666, 2024-01-16:
In your config, you have 
```
          <key>VolumeType</key>
          <string>Internal</string>

```
This means that this custom entry will exist only if the disk is Internal. But the disk is USB, so considered as external.
Just remove these VolumeType and try again.

fiesh, 2024-01-17:
I removed all pairs `VolumeType` / `Interal` from `config.plist`, but the behavior and log stay the same: 
[2024-01-17_12-07_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/13962964/2024-01-17_12-07_BOOTX64.EFI.log)

fiesh, 2024-01-17:
> This means that this custom entry will exist only if the disk is Internal. But the disk is USB, so considered as external. Just remove these VolumeType and try again.

Also, these entries in `config.plist` aren't USB disks but NVMe ones -- just in case that matters.

fiesh, 2024-01-17:
> It also looks like you only have 1 volume online. Is that correct ? Aren't you supposed to have 8 ?

Argh, I had to move `nvmexpressdxe.efi` from `off` to `UEFI`... I did this on the old test installation but not on the new one.

But, anyhow now I have this issue:

It works flawlessly!!! :-)  Thank you so much.  I owe you a beer.

jief666, 2024-01-17:
Ah yes, I should have seen that the USB was your stick, not the actual volume you want to boot on.

It's not finished, the version of Clover you have will not make it to the repo. Will have to make it nicer with a setting in config.plist. So you can have the same version as anyone (I don't want to compile you a new version each time !!).

But first, I'm wondering : why don't you copy your clover folder in the EFI partition of your linux disk ? Do you need this USB stick which is NOT redundant ? Ayway, just wondering.

I will create a setting for you and will ask you to test it. Thanks.

jief666, 2024-01-17:
Change in your config.plist DefaultVolume to FirstAvailable
```
		<key>DefaultVolume</key>
		<string>FirstAvailable</string>
```
Be sure to set a timeout.

Boot with this : [CloverX64-2024-01-17-18-17-52-7e3ffed-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13966502/CloverX64-2024-01-17-18-17-52-7e3ffed-dirty-jief.zip)

If it works, I'll commit it so you can later update to official release.

fiesh, 2024-01-18:
Set the `Timeout` to `15` and `DefaultVolume` to `FirstAvailable` and it booted without problems :-)

jief666, 2024-01-18:
Have you checked that if one volume goes offline, it boots on another one ?

jief666, 2024-01-18:
Great. I'll commit and this will be in the next release.
Can we close this issue ?

fiesh, 2024-01-18:
> Great. I'll commit and this will be in the next release. Can we close this issue ?

Absolutely, thanks for your great work and support!

closed 2024-01-18 by fiesh

jief666, 2024-01-19:
> Having the USB stick does introduce a single point of failure

And there is always the possibility that the OS starts to boot and crash. And that won't be detected by Clover.
But if you have a watchdog system that reboots a computer that doesn't work, we can imagine to create a new setting "Random" instead of "FirstAvailable" to maximize the chances to boot on a spare SSD.

> Dell R930 tries to get you into vendor lock

I was just curious. I understand better now why the USB stick. And yes, a readonly one should not fail often.

fiesh, 2024-01-22:
> And there is always the possibility that the OS starts to boot and crash. And that won't be detected by Clover. But if you have a watchdog system that reboots a computer that doesn't work, we can [...]

This would actually be a really nice feature, having both `FirstAvailable` as well as `Random`!

jief666, 2024-01-22:
You'll probably be the only to use that feature, ever.
So if you tell me that **you** personally prefer that feature, I'll do it. In that case, open a new issue as a feature request. In you prefer FirstAvailable, no need for random.
Let me know.

fiesh, 2024-01-22:
> You'll probably be the only to use that feature, ever. So if you tell me that **you** personally prefer that feature, I'll do it. In that case, open a new issue as a feature request. In you prefer FirstAvailable, no need for random. Let me know.

Then just stick with `FirstAvailable` since that should work equally well for our purposes.  Thank you!