CloverHackyColor/CloverBootloader #650: Latest commit causes weird issues with controls in boot menu.

5T33Z0, 2023-11-24:
The last [commit](https://github.com/CloverHackyColor/CloverBootloader/actions/runs/6959929654) somehow messes up the boot menu. No matter which key I press, it results in Booting the highlighted OS entry!

**Examples**:
-  Windows is pre-selected (left), I click the right arrow to select macOS – boom, Windows starts
- Same thing happens when using the mouse:  Windows is pre-selected (left). I move the mouse cursor to the right to select macOS and click. Boom: Windows bootas.

This happens with the default "Clovy" theme as well.

asepms92, 2023-11-25:
Same here.. my laptop have 3 OS installed, when click F3, F1, F2, arrow left, arrow right, up and down, always booting to Windows, like issues in GUI Clover or source codes refit GUI..

5T33Z0, 2023-11-25:
I also noticed that this commit has a negative impact on WIndows 11 as well for some reason. The trackpad doesn't work when booting with this build.

asepms92, 2023-11-25:
The latest commit, Clover just recognizing Ctrl+Alt+Del and Space

asepms92, 2023-11-25:
[image]
But the trick boot with USB Installer OC, select USB Boot Installer, select Recovery or macOS Partition and Restart to Boot Clover

asepms92, 2023-11-25:
[image]
But the trick of boot with USB Installer OC, select USB Boot Installer, select Recovery or macOS Partition and Restart to Boot Clover

closed 2023-11-26 by 5T33Z0