CloverHackyColor/CloverBootloader #651: Early Kernel panic with latest release. Possibly related to BlockSkywalk Quirk

5T33Z0, 2023-11-27:
After updating Clover on 3 system to this build: https://github.com/CloverHackyColor/CloverBootloader/actions/runs/6996153456

I get an early Kernel Panic on my Laptop that requires **BlockSkywalk** Quirk. Since I don't use this Quirk on my other machines and since someone else has the same issue, I think there might be related to BlockSkywalk.

I've created a debug log:

[2023-11-27_15-30_CLOVERX64.efi.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13477143/2023-11-27_15-30_CLOVERX64.efi.log.zip)

Last version that works on this system is: https://github.com/CloverHackyColor/CloverBootloader/actions/runs/6921325173

5T33Z0, 2024-01-04:
Last commit I tested was this and it worked fine: https://github.com/CloverHackyColor/CloverBootloader/actions/runs/7398572951

jief666, 2024-01-04:
Seems resolved.

closed 2024-01-04 by jief666

LeeBinder, 2024-01-05:
macOS Sonoma booting fine with `BlockSkywalk` Quirk + latest Clover beta from CircleCI, and Broadcom Wi-Fi working fine as already with latest OC.

Thanks @jief666 , well done 👍 

(So yes, the GUI freeze with 0094e0c is definitely a new and separate bug.)