CloverHackyColor/CloverBootloader #653: no idea how to build EFI for lenovo amd laptop..

bfitzgit23, 2023-12-15:
hey guys,

sorry to put this here but, i have no honest idea how to make an EFI for macOS on my laptop. It's an amd lenovo legion 5 17arh05h with 2 SSDs 1tb nvme samsung 980 evo pro 1tb wdc one with win11 one with gentoo linux. 32gb ddr4 3200mhz ram, 8 cores 16 threads ryzen 7 4800h cpu + apu nvidia rtx 2060 dgpu, broadcom bcm94360ng wifi + bluetooth and realtek rtl8111 ethernet, anywho. I have no honest clue how to make clover work on this, i did check the wiki here, still lost as there's tbh only OC info on it and that didn't help me with clover and i DID check the other issues posted im still clueless  so if anyone could please help that'd be great.. Thanks, ben.

SergeySlice, 2023-12-17:
And what is not working? Did you propose to get EFI folder tuned for your computer and for your partitions from someone?

bfitzgit23, 2023-12-17:
No, as I say I don’t understand the clover guide on this repo, whatsoever, not one bit. Nor the OC guide. I basically just followed a YouTube video to make a efi for my laptop but there’s no such thing for clover.

SergeySlice, 2023-12-27:
And what is the issue? That nobody created youtube video?

closed 2023-12-27 by SergeySlice