CloverHackyColor/CloverBootloader #656: [AMD AM5] crackling audio problem

fabiosun, 2024-01-05:
AMD* AM5 platform has serious problem with audio using OSX

Newest AM5 platforms have in their CPU an integrated GPU which is not supported by OSX
We can disable internal GPU in different way but this fact dont solve audio problem

Old Opencore bootloader versions had also this audio crackling problem
Problem was solved using some code integrated in Clover bootloader:
https://github.com/acidanthera/OpenCorePkg/pull/396

Attached config.plist and logs for Clover and Opencore Bootloader:
[bootloaders.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13839045/bootloaders.zip)

* AMD CPU need some kernel patches to boot in OSX, Clover and OpenCore bootloaders can use kernel patches with little differences in their main parameter (missing or only differences in their name):
https://www.insanelymac.com/forum/topic/304530-clover-change-explanations/?do=findComment&comment=2740755

If you need other informations feel free to ask

some links:
audio problem example:
https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2815135

Amd forum where the problem was solved in OpenCore:
https://forum.amd-osx.com/threads/ryzen-7000-testing.3318/page-12#post-22502
https://forum.amd-osx.com/threads/ryzen-7000-testing.3318/page-18#post-22643
https://forum.amd-osx.com/threads/ryzen-7000-testing.3318/page-23#post-22862
https://forum.amd-osx.com/threads/ryzen-7000-testing.3318/page-58#post-24829

jief666, 2024-01-05:
Can you first send me a log of a successful boot with OC ?

jief666, 2024-01-05:
Ah yes, sorry I missed that.
Could you try this :[OpenCore4.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13844427/OpenCore4.efi.zip) ?
NOTE : it should work as before, no modification yet.
Send me the log please.

fabiosun, 2024-01-05:
[opencore-2024-01-05-164720.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13844548/opencore-2024-01-05-164720.txt)
but i cant see any booting drive as you can see in picture below
[image]

fabiosun, 2024-01-05:
this is with original OpenCore.efi
[image]

jief666, 2024-01-05:
I have an clue. Can you send me the ApfsDriverLoader.efi you are using from your OC folder ?

fabiosun, 2024-01-05:
[image]
i use this driver
those driver you say is in clover bootloader driver folder

jief666, 2024-01-05:
I was asking for the driver ApfsDriverLoader.efi from your OC folder.
But I may have found it. Try this : [OpenCore6.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13844833/OpenCore6.efi.zip)

fabiosun, 2024-01-05:
[image]
I do not have those driver in OC folder/driver
I have in Clover/driver

fabiosun, 2024-01-05:
with opencore6.efi i have a black screen insted of bootloader menu
this is the log
[opencore-2024-01-05-173138.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13844914/opencore-2024-01-05-173138.txt)
 ApfsDriverLoader.efi should be deprecated from a bit of time in Opencore (but maybe i am not understanding well your input)

jief666, 2024-01-05:
I didn't know that ApfsDriverLoader.efi wasn't needed anymore !
Looks like your config.plist changed from the one inside bootloaders.zip. It was 67943 and it's now 68768.
Can you make a log with the version 0.9.7 DEBUG of OC ?

Sorry, but it happens to me to search for a problem for 2 days. In the end, it was the user using a wrong config.plist.
I really want to make sure that your are using the exact same OC folder and that you don't modify it between the tests.

Do you use BootloaderChooser so you have a copy of your OC folder dedicated to our tests ?

fabiosun, 2024-01-05:
[opencore-2024-01-05-180216.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13845120/opencore-2024-01-05-180216.txt)
@jief666 i use your bootchooser which helps o much in this tests
in my EFi i have
Clover
OC
OC Copia
Clover is from an Action of Clover Github
OC is my usual Opencore debug (with a proper config for OCLP)
OC Copia is without any modify used for OCLP (OpenCore Legacy Patcher)

fabiosun, 2024-01-05:
by the way, in the log i see always 68768 
i do not know if in bootloader.zip i have unflagged some kext to be "clover" compatible 
I mean some kexts i use in OpenCore does not work well in Clover bootloader

fabiosun, 2024-01-05:
ah ok i know the reason of that bytes difference:
[image]

I added today an entries to try a thing..but it is not relevant because it is disabled :)
Sorry

jief666, 2024-01-05:
I'm very puzzled. The log of the successful boot  is identical as the open generated by OpenCore6.efi (except that it stops at "Ready for takeoff in 0 us", or course).
Are you sure you don't have a cold/warm reset problem ?
Have tried to see if the computer boots anyway ? In other words, is it a freeze or just a black screen but OC is still functioning ?

jief666, 2024-01-05:
And try this : [OpenCore7.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13845856/OpenCore7.efi.zip)

fabiosun, 2024-01-05:
https://github.com/CloverHackyColor/CloverBootloader/assets/6358373/c48fbcae-9580-48ce-8bce-144ec4f7f514

https://github.com/CloverHackyColor/CloverBootloader/assets/6358373/8eba3223-a801-415e-b5e3-927eb6cd7878

https://github.com/CloverHackyColor/CloverBootloader/assets/6358373/f0acb818-fc82-4b49-aaaf-abc0ab142bb6

[opencore7.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13847316/opencore7.txt)
[opencore6.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13847317/opencore6.txt)
[opencore.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13847318/opencore.txt)

with OpenCore 6/7 i have black screen (there is signal on monitor but i have to reset the pc because is blocked)

fabiosun, 2024-01-06:
@jief666 hi and good morning
i did some progresses using opencore4.efi and adding old and deprecated in Opencore bootloader ApfsDriverLoader.efi
System boots and i see all my disks
OSX seems to work as usual (well)
No audio crackling
Config.plist bytes are greater then before because i have added ApfsDriverLoader.efi driver in my config
Attached log, ApfsDriverLoader.efi and OpenCore4.efi

[Archive 2.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13848592/Archive.2.zip)

jief666, 2024-01-06:
In OpenCore that was this problem with the PeCoff library that prevent OC to automatically load apfs.
I fixed that in OpenCore 6 and 7 and I don't understand where the black screen come from.
I have to recreate that bug so I can trace it under gdb, else, I'll have to add log and ask you to do a lot of tests.
Let's first try to reproduce. I'll get back to you.

jief666, 2024-01-06:
I'm thinking of something : could you try in text mode ?
While i'm trying in graphic mode...

fabiosun, 2024-01-06:
I use text mode in clover and graphical mode in OC
put also OC in text mode?

fabiosun, 2024-01-06:
it works :)

fabiosun, 2024-01-06:
[opencore-2024-01-06-112927.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13849437/opencore-2024-01-06-112927.txt)

jief666, 2024-01-07:
Could send me a log made with [CloverX64-2024-01-05-16-57-56-53de33a-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13852956/CloverX64-2024-01-05-16-57-56-53de33a-dirty-jief.zip)
At the beginning of the log, it should be "OC: Starting OpenCore8".

PS : it maybe strange to you that I spend that much time have OC to work, but that's how I plan to find the difference between a Clover boot and an OC boot. And it's better with my OC version because the log will be similar therefore easier to compare.

fabiosun, 2024-01-07:
no, for me it is not strange
I follow your indication because i think you know what is useful or not to achieve "the goal" :)

jief666, 2024-01-07:
> I follow your indication because i think you know what is useful or not to achieve "the goal" :)

Thank your for your trust.

fabiosun, 2024-01-07:
system does nto start
[image]
this is the log i see
[2024-01-07_10-26_CloverX64-2024-01-05-16-57-56-53de33a-dirty-jief.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/13853003/2024-01-07_10-26_CloverX64-2024-01-05-16-57-56-53de33a-dirty-jief.efi.log)

fabiosun, 2024-01-07:
PS i have inserte efi driver in Clover folder
Is it correct?

fabiosun, 2024-01-07:
i also tried it inside OC folder and i see clover inetface with clover logo but only with windows disk
i have added also APFSLoaderDriver kext but no OSX present as before (no log created in both case)

fabiosun, 2024-01-07:
from Clover Log
`FSBFrequency = 52 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
FSB: 100 MHz`
From Opencore log
`00:000 00:000 OCCPU: Failed to get FSBFrequency data using Apple Platform Info - Not Found
00:113 00:000 OCCPU: FSBFrequency   102080764Hz   102MHz
00:198 00:001 OCCPU: Failed to get FSBFrequency data using Apple Platform Info - Not Found
00:204 00:001 OCCPU: Failed to get FSBFrequency data using Apple Platform Info - Not Found
00:210 00:001 OCCPU: Failed to get FSBFrequency data using Apple Platform Info - Not Found
00:766 00:000 OCDH: Setting DataHub 64517CC8-6561-4051-B03C-5964B60F4C7A:FSBFrequency (8) - Success
02:518 00:004 OCAK: Setting FSB and TSC for Family 0xF and ExtFamily 0xA
`
from clover (TSC)
`0:100  0:100  MemLog inited, TSC freq: 4500004050
0:905  0:000  === [ InitScreen ] ==============================
TSC: 4500 MHz
Calibrated TSC Frequency = 4500004050 = 4500MHz`

from Opencore (TSC)
`00:111 00:001 OCCPU: CPUFrequencyFromTSC  4491553630Hz  4491MHz
00:765 00:000 OCDH: Setting DataHub 64517CC8-6561-4051-B03C-5964B60F4C7A:InitialTSC (8) - Success
02:518 00:004 OCAK: Setting FSB and TSC for Family 0xF and ExtFamily 0xA
02:520 00:000 OCAK: TscFreq = 4491553630Hz, TscFCvtt2n = 956231996, TscFCvtn2t = 4491553633`

i put here this diffrences in detected TSC and FSBFrequency

jief666, 2024-01-08:
> system does nto start <img alt="image" width="664" src="https://private-user-images.githubusercontent.com/6358373/294751969-aa3e234f-9f6b-4fd5-bb2e-201bd644f15f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2MjM3NjcsIm5iZiI6MTcwNDYyMzQ2NywicGF0aCI6Ii82MzU4MzczLzI5NDc1MTk2OS1hYTNlMjM0Zi05ZjZiLTRmZDUtYmIyZS0yMDFiZDY0NGYxNWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDdUMTAzMTA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODNmMmNlOWFiZTc3OWRkZTZkNmE2NjE5Zjc2M2RmMWI5N2YwMGM0YzFhNGJlNGQ0ZDYyZDM2NzAyODM2ZjA5MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Bq_pLrFpCIMnMtncgADX4oOP5ayVcgCIY5Px5NQS5YM"> this is the log i see [2024-01-07_10-26_CloverX64-2024-01-05-16-57-56-53de33a-dirty-jief.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/13853003/2024-01-07_10-26_CloverX64-2024-01-05-16-57-56-53de33a-dirty-jief.efi.log)

I think I found that problem. Try : 
[CloverX64-2024-01-08-11-28-32-38564b2-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13858894/CloverX64-2024-01-08-11-28-32-38564b2-jief.zip)

> PS i have inserted efi driver in Clover folder
> Is it correct?

I'm not really sure what that means. If it's about apfs driver, I personally use [apfs.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/13858912/apfs.efi.zip) instead of ApfsDriverLoader.efi

About frequencies :
If you're worrying about difference like 4500004050 and 4491553630Hz, I think it's ok.

I won't be available in the next 24/36 hours. But I'm still on it !

fabiosun, 2024-01-08:
[2024-01-08_10-54_CloverX64-2024-01-08-11-28-32-38564b2-jief.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/13859112/2024-01-08_10-54_CloverX64-2024-01-08-11-28-32-38564b2-jief.efi.log)

System boots correctly (maybe a bit slower but i think for debug efi file you are using
Audio problem is present

> won't be available in the next 24/36 hours. But I'm still on it !

when you have time and if you like to solve this problem :)
Thank you

jief666, 2024-01-09:
> some kexts i use in OpenCore bootloader cant work in Clover Bootloader

You have to explain that because I don't see why it could be the case.

> I have also tested with minimal kexts situation

Minimal situation is good. In tat case, can you make an OC folder and a Clover folder, both in minimal and strictly identical situation and send me the log. Use the exact same kexts. If it doesn't boot, lets find why, because kexts doesn't have any influence on whether the kernel boot or not. At least as far as i know.

I don't have the hardware. So my main chance to find the problem, is to identify a difference when you use Clover vs OC. Which mean we have to remove any differences, as little as they are.

fabiosun, 2024-01-09:
[opencore-2024-01-09-151439.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13875566/opencore-2024-01-09-151439.txt)
[2024-01-09_15-17_CLOVERX64.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/13875567/2024-01-09_15-17_CLOVERX64.efi.log)

> You have to explain that because I don't see why it could be the case.

i think it is about lilu dependances for some kext

then when you will solve this problem about audio we could also evaluate this part :
https://www.insanelymac.com/forum/topic/304530-clover-change-explanations/?do=findComment&comment=2740755
some part are different or missing in the two bootloader
and this part could be important for AMD kernel patches

fabiosun, 2024-01-09:
i did an error using a previous clover efi
i am out my home so i cant provide the same for now (same kexts for the two bootloader with one of your latest jieef debug clover efi)

jief666, 2024-01-09:
I think we should share an EFI folder containing the OC and Clover folder, that way I can have a look at it, make the modification I'd like. Test some of them in VMWare.
And when I'll need a test from you, you'll just have to do "git pull".

To share an EFI folder, have a look at https://jief-machak.gitbook.io/cloverhelp/initial-setup

fabiosun, 2024-01-09:
@jief666 i did an invite but i am not so expert in this field
however in the zip file are present both EFI i use with your bootloader chooser

i f it is not good i hope some others AM5 users could help better to solve this problem

however thank yu for your time ;)

jief666, 2024-01-09:
Sorry, was busy trying to compile secure boot...
We need the repo rather than the zip because I will commit things, you'll be committing things. Sending back and forth end up in mistakes.
I got your invite. Good.
You can confirm that you have your repo IN the boot volume (usually the EFI partition). The point is to NOT have to copy file when I commit. Just a git pull and reboot. Do you confirm ?

fabiosun, 2024-01-09:
> Sorry, was busy trying to compile secure boot... We need the repo rather than the zip because I will commit things, you'll be committing things. Sending back and forth end up in mistakes. I got your [...]

Sorry you mean you modify it in my github and then I download it and  replace / copy it in my EFI folder?

jief666, 2024-01-09:
The point is to NEVER copy any files. This always goes wrong in time.
There was a small. I think you were one folder level too high. No big deal, I've corrected them for you, but you'll to reclone.

But first what is the volume where you will put this ? A usb stick, a second hard drive ?

fabiosun, 2024-01-09:
> But first what is the volume where you will put this ? A usb stick, a second hard drive ?

Problem is I am working on my main volume for this I prefer to copy my self from updated copy of your modified efi to my disk
Otherwise I have to mount my EFI and to modify it directly (I do not like in this way)

fabiosun, 2024-01-09:
Or I have to organize it on an usb bootable EFI?

fabiosun, 2024-01-09:
[image]
i would delete only EFI/clover folder so we would have only the folder shown in the above pictures
Ok?

jief666, 2024-01-09:
There few possibilities.
Yes, You can use another volume (EFI partition of a second SSD, USB, external SSD or HDD).

You can also re-clone that repo directly on your main volume, then add any working OpenCore and/or Clover. As long as the name start with OC, with OpenCore or with Clover, that folder will be ignored. That way, any push or pull won't impact this folder. They won't be sent to github, I won't see them, meaning I can't touch them.

Of course, I understand why you think you can just pull and copy to your main volume each time. In my experience, it happens all the time that after a while, we copy the at the wrong and then I'm searching for hours, not understanding why my modification didn't work...

NOTE : I've already modified the repo, so do a git pull or re-clone anywhere on your SSD.

jief666, 2024-01-09:
> <img alt="image" width="226" src="https://private-user-images.githubusercontent.com/6358373/295345549-4f73d1ac-51b8-430f-a7bd-d7aa04e47d59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4MzY0MTEsIm5iZiI6MTcwNDgzNjExMSwicGF0aCI6Ii82MzU4MzczLzI5NTM0NTU0OS00ZjczZDFhYy01MWI4LTQzMGYtYTdiZC1kN2FhMDRlNDdkNTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDlUMjEzNTExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTExODY4Y2Q3MzRkZjZlODM2MDBiNTNiNGUxZGI5YzU0N2RmYzkwNWYxYTFiYTJlZWRmNDk3NjlhOTQwNzFlNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1kLuWsCUUfOxhhvseD9iBSjVcY2ev_p3NjEuTRqMc_Y"> i would delete only EFI/clover folder so we would have only the folder shown in the above pictures Ok?

No, you still have to rename your EFI folder from your EFI partition. Clone on your main volume " git clone {your repo} EFI" and then add back the OC & Clover folder you'd like.
Then, if you do a git status, you'll see that git will just ignore them.

fabiosun, 2024-01-09:
in my local folder i have this:
[image]

i need of boot/OC/OC COPIA folders (to stay safe)
OC is my main boot folders with OCLP (OpenCore Legacy Patcher applied)
OC Copia is only used to update OSX (it is not possible with regular OC folder for different reasons)

Clover.jief and OC.jief are yours and i will not touch them
i will boot as you suggest with any Jief folders using bootloader chooser
I hope in this way it could work

jief666, 2024-01-09:
OC should be ignored (and anyway I won't touch it).
OC copia is not ignored because you committed it before adding it's name in the .gitignore file. I won't touch it anyway.
I won't touch bootloaderchooser folder named BOOT either.
To be extra safe, you can have a usb stick that allows you to boot. Anyway, you should always have that just in case...

fabiosun, 2024-01-09:
ok thank jief
I am ready! :)

jief666, 2024-01-10:
Ok. So you can pull and reboot on OC.jief/OpenCore9.efi.
Then, you do `cd /Volumes/EFI/EFI && git add -A && git commit -m "fabiosun" && git push` and you send me a message.

fabiosun, 2024-01-10:
sorry no so much skilled here:
`Last login: Wed Jan 10 16:01:59 on console
fabio@Pro-di-fabio-2 ~ % cd /Volumes/EFI/EFI && git add -A && git commit -m "fabiosun" && git push
fatal: not a git repository (or any parent up to mount point /Volumes)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fabio@Pro-di-fabio-2 EFI % 
`
i attached log i see for OC.jief OpenCore9.efi
[opencore-2024-01-10-150007.txt](https://github.com/CloverHackyColor/CloverBootloader/files/13889450/opencore-2024-01-10-150007.txt)

jief666, 2024-01-10:
What the guide told, is to do 
cd /Volumes/EFI
git clone {your repo url} EFI

To check if this worked, you do
cd /Volumes/EFI/EFI
git status

Does that work ?

jief666, 2024-01-10:
BootloaderChoser will allow you to boot from any Clover or OC folder that are in EFI.bak, so you should not be locked out.

jief666, 2024-01-10:
Sorry to insist, but I will have a lot of file to modify so it's a bit slow to setup but way faster afterward.

jief666, 2024-01-10:
Ah yes, I forgot that github change their authentication. You need a personal access token (PAT). See here : https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/managing-your-personal-access-tokens

jief666, 2024-01-10:
> i am able to use github desktop app if it can be useful

Well, yes. works too.

fabiosun, 2024-01-10:
mmmh sorry i did something but it is not my disk EFI folder
it is local copied folder
I am not able to take this from my EFI disk partition

fabiosun, 2024-01-10:
@jief666 i did another invite
this latest invite could be set well with github desktop app which works directly with my disk EFI partition
if not.i am not able to do this as you say is useful (sorry)

jief666, 2024-01-10:
Ok, I got the invite. This is a new repo.
The previous invite did work well : i was able to commit.
Fine let's work with the new one.
NOTE : you didn't the take the .gitignore file, that's why you committed all the ._* files. It's ok.

Can you now configure inside OC.jief so it boots your computer ? Once done, just add everthing and commit.

fabiosun, 2024-01-10:
OC.jief works if i use Opencore.efi
If i use opencore9.efi i see bootmenu without booting OSX disk because it lacks of Driver ApfsDriverLoader.efi (this is deprecated in Opencore as we said before)
If i add ApfsDriverLoader.efi which i have in Clover Driver it boots

So how proceed?
1)
use original Opencore.efi
2) use OpenCore9.efi and add ApfsDriverLoader.efi driver in Opencore config?

jief666, 2024-01-11:
> If i add ApfsDriverLoader.efi which i have in Clover Driver it boots

I'd like to understand why OpenCore9.efi needs ApfsDriverLoader.efi. OpenCore9 is supposed to be the exact 0.9.7 version, just recompiled by myself with few added log.
So lets start to find this bug because it could hides something bigger which could be the cause of other things.

> Opencore log is out of the shared repo because is a level up

Yes, I forgot that Clover put that at top level. I will modify that but for now, just move the log in the EFI folder (down one level) before you git add and commit.

You should have few logs, at least one with OpenCore.efi, and some with OpenCore9.efi with and without ApfsDriverLoader.efi. Commit everything and I'll have a look.

Today and tomorrow and I won't be available at all. This is the last message I'm sending today. I'll get back to it thsi weekend.

fabiosun, 2024-01-11:
Hi Jief
i did as you ask
i added also some comment in fabiosun.txt i post them also here:

`Fabiosun's Comment 

Original OpenCore.efi debug
opencore-2024-01-11-102515.txt
System boots fine

Jief's OpenCore9.efi
opencore-2024-01-11-102754.txt
No OSX drives in OpenCore boot menu

Jief's OpenCore9.efi + ApfsDriverLoader.efi added in config.plist
opencore-2024-01-11-103211.txt
System boots fine`

[48 more replies]